dronefleet/mavlink #10: Might be good for message ID to be accessable at the MavlinkMessage/MavlinkMessage2 level and to have a list of ID's (static or enum) to compare against

Enlade, 2018-09-27:
It seems to me that it would be desirable to make the message ID of the payload available without having to cast; possible make the ID an iVAR of MavlinkMessage/MavlinkMessage2. The ID could be the integer value or more likely you would want to make it an enum.  Either way, the entire list of ID's should be also made available so that some arbitrary message ID can be determined from that list. The most useful reasons for this is so that a user can use a case statement in their message handler method. Message handling is suggested to be done either by using a bunch of "if" statements comparing the class of the received message to each of the known classes (as described in the readme):

```
if (message.getPayload() instanceof Heartbeat) {
	// This is a heartbeat message
	MavlinkMessage<Heartbeat> heartbeatMessage = (MavlinkMessage<Heartbeat>)message;
}
```
But a more interesting way that was suggested was to use some form of pub/sub mechanism, like RxJava2 for example:

```
Observable<Object> observable = PublishSubject.create();

// this is where you publish an incoming message
while ((message = connection.next()) != null) {
    observable.onNext(message.getPayload()); 
}
observable.onComplete();

// this is where you subscribe to messages
observable.ofType(Heartbeat.class)
	.subscribe((heartbeat) -> {
        // do whatever you want with the heartbeat here.
    });
```
And it was suggested that even though a case statement would be more efficient that there are far larger bottlenecks involved that make the case statements impact insignificant.  However, it would still be easier for some users to simply be able to use a case statement to handle messages like:

```
MavlinkConnection connection = MavlinkConnection.create(in, out); // InputStream, OutputStream
MavlinkMessage message;
while ((message = connection.next()) != null) {
	switch (message.Id) { //<=== No such IVar so cannot do this
		case MAVLINK_MSG_ID_HEARTBEAT : { //MAVLINK_MSG_ID_HEARTBEAT = 0
			//handle heartbeat
			Heartbeat tempmsg = (Heartbeat) message.getPayload();
			...
			...
			} break;
		case MAVLINK_MSG_ID_SYS_STATUS : { //MAVLINK_MSG_ID_SYS_STATUS = 1
			//handle system status
			} break;
		...
		...
		...
		case MAVLINK_MSG_ID_ATTITUDE_CONTROL : { //MAVLINK_MSG_ID_ATTITUDE_CONTROL = 200
			//handle attitude control
			} break;
		default: {
			System.out.println("MSG Error : unknown MsgId (" + myMsgID + ")");
			}
	}
}
```
There are other areas of the code that might also be made easier to implement if the message ID and list of message IDs were available. For instance, when creating a hierarchical list box of all incoming messages the message ID could be used as the unique identifier to determine whether a new message is to be added to the hierarchy or if an existing message needs to be updated.  All of these uses have work arounds, but if it easy enough to add this extra functionality then maybe it would be useful to do so, if not to simply allow the users to code their projects in whichever way they are most comfortable.

benbarkay, 2018-09-28:
You can obtain a message's ID from reading the annotations:

```java
@Test
public void test() throws InterruptedException {
    CommandLong message = CommandLong.builder()
            .command(MavCmd.MAV_CMD_COMPONENT_ARM_DISARM)
            .param1(1)
            .param2(2989.0f)
            .targetSystem(1)
            .targetComponent(0)
            .build();

    MavlinkMessageInfo messageInfo = message.getClass().getAnnotation(MavlinkMessageInfo.class);
    System.out.println(messageInfo.id());
}
```

Enlade, 2018-09-28:
Yea, I was thinking something like:

`msgID = message.ID();
`

might be easier for people is all.

benbarkay, 2018-09-28:
This library isn't built for people who are learning Java, it's designed and built for developing software in Java, idiomatically, and as such utilizes the language's type-system for distinguishing between types, and annotations to store meta information about these types -- a practice that is common across the board with other popular libraries such as Jackson, Hibernate, Spring and more.

I am sorry but the trade you are proposing isn't one I'm willing to make. I believe that you'll find it rewarding and worthwhile to spend some time to understand how this all works, though.

closed 2018-10-03 by benbarkay