dronefleet/mavlink #14: Testing Mavlink 2

kgmormylo, 2018-10-30:
Hi all. 
I try to use mavlink2.
When i create and send mavlink2 message i receive error with signing package.

// Now we are ready to read and send messages.
MavlinkMessage message;
while ((message = connection.next()) != null) {
		// When a message is received, its payload type isn't statically available.
		// You can resolve which kind of message it is by its payload, like so:
		if (message.getPayload() instanceof Heartbeat) {
			// This is a heartbeat message
					System.out.println("Heartbeat");
					System.out.println(message.toString());
					if ( handshake != true ) {
						
						System.out.println("\n Handshake \n ");

						connection.send(new Mavlink2Message<>(
								1,
								1,
								message.getOriginComponentId(), 
								message.getOriginSystemId(), 
								CommandInt.builder()
								.targetComponent(message.getOriginComponentId())
								.targetSystem(message.getOriginSystemId()).param1(1)
								.frame(MavFrame.MAV_FRAME_MISSION)
								.command(MavCmd.MAV_CMD_REQUEST_PROTOCOL_VERSION)
								.build()
						      ));
						handshake = true;						
					}
                    }
}

kgmormylo, 2018-10-31:
Sorry, Ben. I edit message and add exception stack to it. But it didn't save. 

```
Exception in thread "main" java.lang.IllegalStateException: Failed generate signature for packet
	at io.dronefleet.mavlink.protocol.MavlinkPacket.sign(MavlinkPacket.java:298)
	at io.dronefleet.mavlink.MavlinkConnection.send(MavlinkConnection.java:327)
	at simpleMavlink.Main.main(Main.java:157)
Caused by: java.security.DigestException: Length must be at least 32 for SHA-256digests
	at sun.security.provider.DigestBase.engineDigest(Unknown Source)
	at java.security.MessageDigest$Delegate.engineDigest(Unknown Source)
	at java.security.MessageDigest.digest(Unknown Source)
	at io.dronefleet.mavlink.protocol.MavlinkPacket.sign(MavlinkPacket.java:290)
	... 2 more
```

Enlade, 2018-10-31:
I think you need to do a bunch more to set up the messages right? Like you are hitting a signing issue because you didn't set up the signatureConfiguration.  I think there are other things you need to set up as well.  Again, I just peaked at V2 so I didn't look into it too closely, but my understanding is that you need to do more with V2 than you would with V1 (just calling send()).

Enlade, 2018-10-31:
You probably did some of this in the code you didn't provide that opens the connection.  I see in the readme file that you need to set up the signing and other stuff in the following way (this is for sockets, were I am using serial, but it should be about the same).  Do you do this?

```
    MavlinkConnection connection = MavlinkConnection.builder(socket.getInputStream(), socket.getOutputStream())
            
            // Dialects are associated to systems when a heartbeat is received. The heartbeat
            // informs us of the autopilot of the device, which tells us which dialect
            // that device adheres to.
            .dialect(MavAutopilot.MAV_AUTOPILOT_GENERIC, new StandardDialect())
            .dialect(MavAutopilot.MAV_AUTOPILOT_ARDUPILOTMEGA, new ArdupilotmegaDialect())
            
            // When specifying signing configuration, every Mavlink2Message that is send
            // through this connection will be signed. We will need to setup signing
            // by sending Mavlink1 messages before we can communicate with Mavlink2 messages.
            .signing(new SigningConfiguration(
                    0, // This is the initial timestamp for signing, We should only specify
                                // a value other than 0 if we do not trust that our system's clock is
                                // going to be calibrated prior to the connection. Generally, this value
                                // is loaded from persistence (where the last signature timestamp is stored)

                    1,    // The signing link ID. Generally, this should be connection-bound. So if there
                                // is more than a single connection, each of them has to have a different value.

                    // Secret key, should be 36 bytes in length. The specification recommends using a
                    // SHA-256'd passphrase.
                    MessageDigest.getInstance("SHA-256")
                            .digest("my secret key".getBytes(StandardCharsets.UTF_8))
            ))
            .build();
```

kgmormylo, 2018-10-31:
Enlade, i copy example and after that add my code that you could see here in while loop

Enlade, 2018-10-31:
The key in the example "my secret key" isn't 36 bytes in length.  So, maybe try changing it to something that is 36 in length if you haven't done that already?

kgmormylo, 2018-11-09:
Hey, guys. Do you have any ideas? When i commend sing method call error disappeared.

benbarkay, 2018-11-09:
It looks like there's something else with signing that isn't right. If you can do without signing for a while then I'll be able to properly test and fix this within a couple of weeks.

kgmormylo, 2018-11-12:
thanks, Ben. i am looking forward in update. If you need any help write me and good luck.

commit b696e121 (2018-11-13): fix: #14
closed 2018-11-13 by benbarkay