dronefleet/mavlink #15: Problem with mavlink2 deserialization

kgmormylo, 2018-10-30:
I'v met another problem.
I plug my APM. APM start send me message. I see that is mav2 message because i see method From2 of MavlinkPacket work.
But after some portion of message program terminated with error.

```
Start
fromV2
MavlinkPacket constructor
fromV2
MavlinkPacket constructor
fromV2
MavlinkPacket constructor
Exception in thread "main" java.lang.ArrayIndexOutOfBoundsException
	at java.lang.System.arraycopy(Native Method)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.lambda$5(ReflectionPayloadDeserializer.java:57)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(Unknown Source)
	at java.util.stream.ReferencePipeline$2$1.accept(Unknown Source)
	at java.util.ArrayList.forEach(Unknown Source)
	at java.util.stream.SortedOps$RefSortingSink.end(Unknown Source)
	at java.util.stream.Sink$ChainedReference.end(Unknown Source)
	at java.util.stream.AbstractPipeline.copyInto(Unknown Source)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(Unknown Source)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(Unknown Source)
	at java.util.stream.AbstractPipeline.evaluate(Unknown Source)
	at java.util.stream.ReferencePipeline.forEach(Unknown Source)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.deserialize(ReflectionPayloadDeserializer.java:48)
	at io.dronefleet.mavlink.MavlinkConnection.next(MavlinkConnection.java:272)
	at simpleMavlink.Main.main(Main.java:110)
```

benbarkay, 2018-10-30:
Looks like the field is missing in the received packet, perhaps it is an extension field. It would be helpful if you could save `payload` bytes to a file and upload it here so that I can reproduce this. Also, what's the output of `messageType.getSimpleName()`?

kgmormylo, 2018-10-31:
As i mentioned it before it doesn't depend of message type. but i add system.print of getSimpleName.

First attepmt

`
Start
fromV2
MavlinkPacket constructor
Message type simple name ActuatorControlTarget
MavlinkMessage{originSystemId=1, originComponentId=1, payload=ActuatorControlTarget{timeUsec=null, groupMlx=0, controls=null}}
The file has been written
fromV2
MavlinkPacket constructor
Message type simple name Altitude
MavlinkMessage{originSystemId=1, originComponentId=1, payload=Altitude{timeUsec=null, altitudeMonotonic=-45.52188, altitudeAmsl=-45.52188, altitudeLocal=0.5233091, altitudeRelative=0.5233091, altitudeTerrain=NaN, bottomClearance=NaN}}
The file has been written
fromV2
MavlinkPacket constructor
Message type simple name AttitudeTarget
MavlinkMessage{originSystemId=1, originComponentId=1, payload=AttitudeTarget{timeBootMs=5957304, typeMask=0, q=null, bodyRollRate=-0.15092327, bodyPitchRate=-0.097558685, bodyYawRate=4.8780342E-4, thrust=0.0}}
The file has been written
fromV2
MavlinkPacket constructor
Message type simple name EstimatorStatus
Exception in thread "main" java.lang.ArrayIndexOutOfBoundsException
	at java.lang.System.arraycopy(Native Method)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.lambda$5(ReflectionPayloadDeserializer.java:62)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(Unknown Source)
	at java.util.stream.ReferencePipeline$2$1.accept(Unknown Source)
	at java.util.ArrayList.forEach(Unknown Source)
	at java.util.stream.SortedOps$RefSortingSink.end(Unknown Source)
	at java.util.stream.Sink$ChainedReference.end(Unknown Source)
	at java.util.stream.AbstractPipeline.copyInto(Unknown Source)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(Unknown Source)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(Unknown Source)
	at java.util.stream.AbstractPipeline.evaluate(Unknown Source)
	at java.util.stream.ReferencePipeline.forEach(Unknown Source)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.deserialize(ReflectionPayloadDeserializer.java:48)
	at io.dronefleet.mavlink.MavlinkConnection.next(MavlinkConnection.java:273)
	at simpleMavlink.Main.main(Main.java:109)
`
payload

`AttitudeTarget{timeBootMs=5957304, typeMask=0, q=null, bodyRollRate=-0.15092327, bodyPitchRate=-0.097558685, bodyYawRate=4.8780342E-4, thrust=0.0}`

***************************************

Second attempt

`Start
fromV2
MavlinkPacket constructor
Message type simple name ActuatorControlTarget
MavlinkMessage{originSystemId=1, originComponentId=1, payload=ActuatorControlTarget{timeUsec=null, groupMlx=0, controls=null}}
The file has been written
fromV2
MavlinkPacket constructor
Message type simple name Attitude
MavlinkMessage{originSystemId=1, originComponentId=1, payload=Attitude{timeBootMs=6068099, roll=0.023623532, pitch=0.015008304, yaw=0.46602818, rollspeed=-6.837901E-4, pitchspeed=-3.273189E-4, yawspeed=8.254666E-4}}
The file has been written
fromV2
MavlinkPacket constructor
Message type simple name HighresImu
MavlinkMessage{originSystemId=1, originComponentId=1, payload=HighresImu{timeUsec=null, xacc=0.12826285, yacc=-0.24170168, zacc=-9.960193, xgyro=-6.837901E-4, ygyro=-3.273189E-4, zgyro=8.254666E-4, xmag=0.088879995, ymag=-0.03616, zmag=0.46008, absPressure=101860.0, diffPressure=0.0, pressureAlt=-44.4491, temperature=42.28, fieldsUpdated=7167}}
The file has been written
fromV2
MavlinkPacket constructor
Message type simple name ServoOutputRaw
MavlinkMessage{originSystemId=1, originComponentId=1, payload=ServoOutputRaw{timeUsec=1773311165, port=1, servo1Raw=1500, servo2Raw=1500, servo3Raw=1500, servo4Raw=1500, servo5Raw=0, servo6Raw=0, servo7Raw=0, servo8Raw=0, servo9Raw=0, servo10Raw=0, servo11Raw=0, servo12Raw=0, servo13Raw=0, servo14Raw=0, servo15Raw=0, servo16Raw=0}}
[... 81 lines cut ...]
MavlinkPacket constructor
Message type simple name Ping
Exception in thread "main" java.lang.ArrayIndexOutOfBoundsException
	at java.lang.System.arraycopy(Native Method)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.lambda$5(ReflectionPayloadDeserializer.java:62)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(Unknown Source)
	at java.util.stream.ReferencePipeline$2$1.accept(Unknown Source)
	at java.util.ArrayList.forEach(Unknown Source)
	at java.util.stream.SortedOps$RefSortingSink.end(Unknown Source)
	at java.util.stream.Sink$ChainedReference.end(Unknown Source)
	at java.util.stream.AbstractPipeline.copyInto(Unknown Source)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(Unknown Source)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(Unknown Source)
	at java.util.stream.AbstractPipeline.evaluate(Unknown Source)
	at java.util.stream.ReferencePipeline.forEach(Unknown Source)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.deserialize(ReflectionPayloadDeserializer.java:48)
	at io.dronefleet.mavlink.MavlinkConnection.next(MavlinkConnection.java:273)
	at simpleMavlink.Main.main(Main.java:109)`

payload

`LocalPositionNed{timeBootMs=6172784, x=0.0, y=0.0, z=-0.9394648, vx=-0.002111224, vy=-4.05088E-4, vz=-0.62612253}`

It seems that problem in offset determining in deserealization class. After some attempts. offset have been determined wrong. if you look at my previous message you'll see that something wrong when system attempt determine offset for arraycopy command.

kgmormylo, 2018-10-31:
> Looks like the field is missing in the received packet, perhaps it is an extension field. It would be helpful if you could save `payload` bytes to a file and upload it here so that I can reproduce this. Also, what's the output of `messageType.getSimpleName()`?

kgmormylo, 2018-10-31:
As i mentioned it before it doesn't depend of message type. but i add system.print of getSimpleName.

First attepmt

```
Start
fromV2
MavlinkPacket constructor
Message type simple name Attitude
MavlinkMessage{originSystemId=1, originComponentId=1, payload=Attitude{timeBootMs=7331520, roll=0.024159597, pitch=0.013583231, yaw=0.45652345, rollspeed=-8.4632076E-5, pitchspeed=1.141876E-4, yawspeed=9.2970766E-4}}
The file has been written
fromV2
MavlinkPacket constructor
Message type simple name HighresImu
MavlinkMessage{originSystemId=1, originComponentId=1, payload=HighresImu{timeUsec=null, xacc=0.14949717, yacc=-0.23806155, zacc=-9.9851465, xgyro=-8.4632076E-5, ygyro=1.141876E-4, zgyro=9.2970766E-4, xmag=0.0892, ymag=-0.03504, zmag=0.46383998, absPressure=101871.0, diffPressure=0.0, pressureAlt=-45.35806, temperature=42.329998, fieldsUpdated=7167}}
The file has been written
fromV2
MavlinkPacket constructor
Message type simple name ServoOutputRaw
MavlinkMessage{originSystemId=1, originComponentId=1, payload=ServoOutputRaw{timeUsec=3036665725, port=1, servo1Raw=1500, servo2Raw=1500, servo3Raw=1500, servo4Raw=1500, servo5Raw=0, servo6Raw=0, servo7Raw=0, servo8Raw=0, servo9Raw=0, servo10Raw=0, servo11Raw=0, servo12Raw=0, servo13Raw=0, servo14Raw=0, servo15Raw=0, servo16Raw=0}}
The file has been written
fromV2
MavlinkPacket constructor
Message type simple name ActuatorControlTarget
MavlinkMessage{originSystemId=1, originComponentId=1, payload=ActuatorControlTarget{timeUsec=null, groupMlx=0, controls=null}}
The file has been written
fromV2
MavlinkPacket constructor
Message type simple name Attitude
MavlinkMessage{originSystemId=1, originComponentId=1, payload=Attitude{timeBootMs=7331768, roll=0.024161302, pitch=0.013563508, yaw=0.4566598, rollspeed=2.4250522E-4, pitchspeed=-1.7407909E-4, yawspeed=-2.4278276E-4}}
The file has been written
fromV2
MavlinkPacket constructor
Message type simple name AttitudeQuaternion
MavlinkMessage{originSystemId=1, originComponentId=1, payload=AttitudeQuaternion{timeBootMs=7331768, q1=0.9739709, q2=0.010231617, q3=0.009339545, q4=0.22624956, rollspeed=2.4250522E-4, pitchspeed=-1.7407909E-4, yawspeed=-2.4278276E-4}}
The file has been written
fromV2
MavlinkPacket constructor
fromV2
MavlinkPacket constructor
Message type simple name Vibration
MavlinkMessage{originSystemId=1, originComponentId=1, payload=Vibration{timeUsec=null, vibrationX=8.291815E-10, vibrationY=6.309114E-6, vibrationZ=3.6353746E-4, clipping0=0, clipping1=0, clipping2=0}}
The file has been written
fromV2
MavlinkPacket constructor
Message type simple name LocalPositionNed
MavlinkMessage{originSystemId=1, originComponentId=1, payload=LocalPositionNed{timeBootMs=7331768, x=0.0, y=0.0, z=-0.8397414, vx=-0.0038822934, vy=0.0056522037, vz=-0.6917211}}
The file has been written
fromV2
MavlinkPacket constructor
Message type simple name ServoOutputRaw
MavlinkMessage{originSystemId=1, originComponentId=1, payload=ServoOutputRaw{timeUsec=3036801501, port=1, servo1Raw=1500, servo2Raw=1500, servo3Raw=1500, servo4Raw=1500, servo5Raw=0, servo6Raw=0, servo7Raw=0, servo8Raw=0, servo9Raw=0, servo10Raw=0, servo11Raw=0, servo12Raw=0, servo13Raw=0, servo14Raw=0, servo15Raw=0, servo16Raw=0}}
The file has been written
fromV2
MavlinkPacket constructor
Message type simple name BatteryStatus
MavlinkMessage{originSystemId=1, originComponentId=1, payload=BatteryStatus{id=0, batteryFunction=EnumValue{value=1, entry=MAV_BATTERY_FUNCTION_ALL}, type=EnumValue{value=1, entry=MAV_BATTERY_TYPE_LIPO}, temperature=32767, voltages=null, currentBattery=-1, currentConsumed=-1, energyConsumed=-1, batteryRemaining=-1, timeRemaining=0, chargeState=null}}
The file has been written
fromV2
MavlinkPacket constructor
Exception in thread "main" java.lang.ArrayIndexOutOfBoundsException
	at java.lang.System.arraycopy(Native Method)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.lambda$5(ReflectionPayloadDeserializer.java:62)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(Unknown Source)
	at java.util.stream.ReferencePipeline$2$1.accept(Unknown Source)
	at java.util.ArrayList.forEach(Unknown Source)
	at java.util.stream.SortedOps$RefSortingSink.end(Unknown Source)
	at java.util.stream.Sink$ChainedReference.end(Unknown Source)
	at java.util.stream.AbstractPipeline.copyInto(Unknown Source)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
[... 397 lines cut ...]
				// We are likely better off by publishing the payload to a pub/sub mechanism
				// such
				// as RxJava, JMS or any other favorite instead, though.

			}
			serialPort.close();
			System.out.println("Stop");
		} catch (SerialPortException spE) {
			System.out.println("SerialPortException");
		} catch (EOFException eofE) {
			// Stream has ended
			System.out.println("EOF Exception");
		} catch (IOException ioE) {
			System.out.println(" IO Exception");

		} catch (NoSuchAlgorithmException e) {
			// TODO Auto-generated catch block
			e.printStackTrace();
		} finally {
			serialPort.close();
		}

	}
}
```

Enlade, 2018-10-31:
Are you sure those are the messages that it is crashing on?  Seems like those are the last successful messages right?  I would be interested in seeing what the actual data is that it is crashing on.  Maybe drop some code in before the arraycopy in ReflectionPayloadDeserializer.java to output some of the relevant information about the actual last message on which the exception occurs?  Like drop some code in after the declaration of AtomicInteger, something like:

Stream<Method> myms = Arrays.stream(builder.getClass().getMethods())
                    .filter(m -> m.isAnnotationPresent(MavlinkFieldInfo.class))
                    .sorted((a, b) -> {
                        MavlinkFieldInfo fa = a.getAnnotation(MavlinkFieldInfo.class);
                        MavlinkFieldInfo fb = b.getAnnotation(MavlinkFieldInfo.class);

                        return MavlinkFieldInfo.WIRE_COMPARATOR.compare(fa, fb);
                    })
                    .filter(f -> nextOffset.get() < payload.length);

            myms.forEach(mym-> {
                MavlinkFieldInfo field = mym.getAnnotation(MavlinkFieldInfo.class);

                int length = field.unitSize() * Math.max(field.arraySize(), 1);
                int offset = nextOffset.getAndAccumulate(length, (a, b) -> a + b);

            	System.out.println("[" + length + ", " + offset + "]" + mym.toString());
            });

Might be some bugs in this code cause I just copy pasted some stuff together and didn't run it.  But you could also output other stuff there as well to help figure out what is different about those messages on which it is failing (at least you will find out which message it is actually failing on).

kgmormylo, 2018-10-31:
Enlade, thanks for reply.
If you look end of my message. there are two log. One in byte has 4 message with following byte array.
Filled in ReflectionPayloadDeserializer.java .
And message log from   main class in while loop.
This files has 3 message.
My stream crash on Ping message in that case(but itsn't depend of message type as i exlpored).
If APM send me V1 message all fine. But then it try to send V2 message you could see it from sytem print. After some message programm crash. I only copy example from github, add serial port stream library and main class append to my message here.

kgmormylo, 2018-10-31:
> Are you sure those are the messages that it is crashing on? Seems like those are the last successful messages right? I would be interested in seeing what the actual data is that it is crashing on. [...]

I already post this output in begin of issue.

```
MavlinkMessage{originSystemId=1, originComponentId=1, payload=AttitudeTarget{timeBootMs=7214174, typeMask=0, q=null, bodyRollRate=-0.114282124, bodyPitchRate=-0.912515, bodyYawRate=0.0034739806, thrust=0.0}}
fromV2
MavlinkPacket constructor
payload [B@433c675d
offset 0
data [B@3f91beef
length 8
payload [B@433c675d
offset 8
data [B@1a6c5a9e
length 4
payload [B@433c675d
offset 12
data [B@37bba400
length 4
payload [B@433c675d
offset 16
data [B@179d3b25
length 4
payload [B@433c675d
offset 20
data [B@254989ff
length 4
payload [B@433c675d
offset 24
data [B@5d099f62
length 4
payload [B@433c675d
offset 28
data [B@37f8bb67
length 4
payload [B@433c675d
offset 32
data [B@49c2faae
length 4
payload [B@433c675d
offset 36
data [B@20ad9418
length 4
payload [B@433c675d
offset 40
data [B@31cefde0
length 2
Exception in thread "main" java.lang.ArrayIndexOutOfBoundsException
	at java.lang.System.arraycopy(Native Method)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.lambda$5(ReflectionPayloadDeserializer.java:62)
```

Enlade, 2018-10-31:
You could also drop in some code to take a quick peak at the payload if you want.  At the beginning of that deserialize method you could put something like:

`System.out.println(String.format("%X", new BigInteger(1, payload)));
`

Note that this effectively prints out the nibbles, so the first nibble will be dropped if it is a 0.  So a heartbeat message will be something like "2 03 51 03 03" instead of "02 03 51 03 03".  Mind you it would probably be more useful to print out the message ID and the payload with it, but you don't have message ID in the deserialize method.  What I would do is to print the message information out in MavlinkConnection.java just before the call to the desirialize, before:

`Object payload = deserializer.deserialize(packet.getPayload(), messageType);
`

Then you can print out something like:

`System.out.println("=========== [" + packet.getMessageId() + "] " + String.format("%X", new BigInteger(1, packet.payload)));
`

Again, I didn't test any of this code above, so there is likely syntax errors or such, but you can see where I'm going here.  That with the other debug code should give you a clear picture of the issue.  Hope that helps.

Enlade, 2018-10-31:
So, can you tell me what the message type is (what is the message ID) for the message it is failing on in the data that you posted?

Enlade, 2018-10-31:
I'm having trouble figuring that out from the data you are posting.  Its probably there, but I'm having trouble is all.  What do you think the message ID is of the message that is failing from the data you just posted?

Enlade, 2018-10-31:
Like, I see that you print out information about the attitude message (or some such).  Then after that you run into the error.  So, doesn't that mean that the attitude message was the previous message...one that succeeded?  So, arn't we interested in the current message that is failing?  What is the current message that failed after you got a successful attitude message?  It might be that I'm just reading your data wrong, so I'm just thinking we are missing the actual message that it is failing on.  Where do you see that in the data you are posting above?

Enlade, 2018-10-31:
To put it another way.  When you print out the attitude message it must already have all of its fields read from the stream.  So, after you print it out and you then start printing out all of the offset/length messages, then they can't be getting the attitude fields cause you already had them when you printed out the attitude message, right?  So, what message do all of those subsequent offset/length fields belong to is what I'm confused about.  I think you don't get to see that because the error occurs before you print out the next successful message (like you did with the attitude message).  So, I was trying to put in some debug code to get that information, but again, maybe I'm just reading your data wrong.  Just tossing out some ideas that may help.

kgmormylo, 2018-10-31:
Ben, this file with byte array and name of message.
[byte_log.txt](https://github.com/dronefleet/mavlink/files/2534983/byte_log.txt)

Enlade, 2018-10-31:
Yea, I hit some issues with V2 messages as well, but I got busy with other things and put that on hold.  Eventually I'll have to revisit them, so, I'm happy to see you working on that effort so maybe you will save me some time :)

In any case, I'll put an extra set of eyes on it and see if I can see anything fishy.

Enlade, 2018-10-31:
Oh, and maybe you could make use of that silly little string to hexstring code that I posted for dumping out the message as a hex string?  It's not efficient or anything, but it is easy to toss in so you don't have to deal with the unprintable characters.

Enlade, 2018-10-31:
Ahhhh, from that file it tells me you are failing on the HighresImu message.  Got it.

Enlade, 2018-10-31:
Or maybe not.  The HighresImu should be looking for length 4 at offset 40 (and not length 2).  So, maybe it is some other message altogether.  I still think you need to toss in some code to print the messageID() of the payload before the error occurs, just to be sure (in MavlinkConnection.java).  Just so we can be sure what message exactly it is failing on (or maybe what message it thinks it is failing on).

Enlade, 2018-10-31:
I did a quick review of the mavlink messages and can't find any that fit that profile of offset/lengths (8, 4, 4, 4, 4, 4, 4, 4, 4, 2).  So, getting that messageID would probably be the best first step.

kgmormylo, 2018-10-31:
Enlade, let's look at my issue one else.
I add system print. before error. after last portion of byte in file error occured.
It is happen in arraycopy command.

```
Arrays.stream(builder.getClass().getMethods()).filter(m -> m.isAnnotationPresent(MavlinkFieldInfo.class))
					.sorted((a, b) -> {
						MavlinkFieldInfo fa = a.getAnnotation(MavlinkFieldInfo.class);
						MavlinkFieldInfo fb = b.getAnnotation(MavlinkFieldInfo.class);
						return wireComparator.compare(fa, fb);
					}).filter(f -> nextOffset.get() < payload.length).forEach(method -> {
						MavlinkFieldInfo field = method.getAnnotation(MavlinkFieldInfo.class);
						int length = field.unitSize() * Math.max(field.arraySize(), 1);

						int offset = nextOffset.getAndAccumulate(length, (a, b) -> a + b);

						byte[] data = new byte[length];
						byte[] buffer = payload.clone();
						try {

							fos.write(buffer, 0, buffer.length);
						} catch (IOException ioE) {
							System.out.println(" IO Exception");
							ioE.printStackTrace();
						}

						System.arraycopy(payload, offset, data, 0, length);
```
that my code in ReflectionPayloadDeserializer class. 
I am newby in Java and have no idea why it's happen. Only one thing that i could suggest that offset is wrong

Enlade, 2018-11-01:
Yes, it is definitely happening with the arraycopy.  My interest is in trying to figure out specifically which message the error is happening on.  There are a lot of message source code files and if one has some bug in it that causes the annotations to be extracted incorrectly or some such then that is what we are after.  Only, I tried to look through to find out what message it was based on what  you printed out for the offset/length messages, but I could not match it up with a message.  Its just easier for you to toss in a little command that prints the message ID, but you can't do that in the deserializer code.  Mind you, I also wanted to see more information coming from the deserializer (like you have already done with printing the offsets and lengths).

In any case, the code I suggested above would tell you that stuff.  Also, that code for the deserializer doesn't change any of the existing code.  I am just repeating the effort, but I don't like all the builder stuff because it makes debugging a bit tedious (not a criticism, the builder is fine).  So, I broke out the code up to the error and then I looped through the methods looking at everything.  But again, you can't see the ID in the deserializer because it is not part of that payload.  So, I also put in a simple one line println to dump out the ID (and I printed the hex string payload there as well) for the message before it calls into the deserializer (in MavlinkConnection.java).  Maybe revert the code back to the way it was in those two source files and then toss in the code I posted above in those two locations (debug a bit cause I didn't test any of that code).  I think it will give you at least a good idea of where the problem is.

kgmormylo, 2018-11-01:
Dear friend thanks for your helping but please look at my messages one else closely and don't hurry up. I repeat several time that error doesn't depend of message. And show some example of that in several messages. It doesn't make sense suggest why it's happen without analyzing code of deserealizing and deep understanding how it works. Problem in offset determination before arraycopy but i couldn't understand algorithm and ask author help .

Enlade, 2018-11-01:
At the very least, just add that one line of code to the MavlinkConnection.java file (io/dronefleet/mavlink/MavlinkConnection.java).  So, under the next() method find the call to the deserializer.  The line of code looks like this:

`            Object payload = deserializer.deserialize(packet.getPayload(), messageType);
`

Its at line 275 or there abouts.  So, just put this line of code right before that line of code.

`            System.out.println("===================================" + packet.getMessageId());
`

So the code should now look like this:

```
            System.out.println("===================================" + packet.getMessageId());
            Object payload = deserializer.deserialize(packet.getPayload(), messageType);
```

I think that that will at least tell us what message it is failing on.  When you have that ID number then just tell us what it is.  The ID of the message that is failing will be shown as the last of these println's before it fails.  I have also shown the HEX dump of the payload so it is more readable and that is also useful.

Now if you want to be complete so that we know which annotation it is failing on then you can also drop in my other code suggestion above.  It might also be useful once we know which message it is failing on to also know what particular method annotation is causing the trouble.  Just suggesting a few things that might help to narrow down the problem.

Enlade, 2018-11-01:
Its just one line of code.  Humor me at least.  Its so easy to just toss it in and post the results.

Enlade, 2018-11-01:
Lets look at your post, please tell me what message this failed on:

```
MavlinkMessage{originSystemId=1, originComponentId=1, payload=AttitudeTarget{timeBootMs=7214174, typeMask=0, q=null, bodyRollRate=-0.114282124, bodyPitchRate=-0.912515, bodyYawRate=0.0034739806, thrust=0.0}}
fromV2
MavlinkPacket constructor
payload [B@433c675d
offset 0
data [B@3f91beef
length 8
payload [B@433c675d
offset 8
data [B@1a6c5a9e
length 4
payload [B@433c675d
offset 12
data [B@37bba400
length 4
payload [B@433c675d
offset 16
data [B@179d3b25
length 4
payload [B@433c675d
offset 20
data [B@254989ff
length 4
payload [B@433c675d
offset 24
data [B@5d099f62
length 4
payload [B@433c675d
offset 28
data [B@37f8bb67
length 4
payload [B@433c675d
offset 32
data [B@49c2faae
length 4
payload [B@433c675d
offset 36
data [B@20ad9418
length 4
payload [B@433c675d
offset 40
data [B@31cefde0
length 2
Exception in thread "main" java.lang.ArrayIndexOutOfBoundsException
	at java.lang.System.arraycopy(Native Method)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.lambda$5(ReflectionPayloadDeserializer.java:62)

```

Enlade, 2018-11-01:
If you say that it failed on the attitude message then I am very confused.  How can you print out the fields of the attitude message before those fields have been deserialized from the stream?  Its not possible.  So, that can't be the attitude message.  The fact that you are showing a call to the message packet constructor makes it clear that that first line (which prints out the attitude) is for the previous successful message and not for the actual message that the arraycopy is failing on.  Right?

All of your other sample posts are similarly unclear.  If the message prints with all of its fields then it has to have already been deserialized from the stream, which means all of them are previous messages.  The message we want is the one that is currently being deserialized, but that crashes before it has been deserialized (which means you cannot print out the fields, cause it didn't get deserialized).  Right?

So, what I am suggesting is that from your posts it seems like you still do not know exactly which message is causing the problem.  Yes, it is happening in the arraycopy and it is happening because the offset/length's are not correct, but what I am saying is to discover why that is the case it would help to get a better look at what message the code is actually failing on.  I mean, how can it hurt to know what the message ID is?

kgmormylo, 2018-11-01:
Look at my second message in this issue. I write free different example of starting programm. Also Ben ask me add this method messageType.getSimpleName(). your could see result of this system.print.
As that way i could research that it doesn't depend of message type.
Now you ask me add code that i already add before in other form only. That's i try to tell you. 
Add code it's not problem.

kgmormylo, 2018-11-01:
when you asked me print payload offset and so on i only answer that i already do it.

Enlade, 2018-11-01:
yes, what you added for printing the offset/length is fine.  That gives us some nice information about where in the current message the error is occurring.  I kind of like to see the hex dump of the payload, but it is ok.  My problem is that I don't know what actual message it is failing on.

Enlade, 2018-11-01:
Ok, so you say that you want me to look at your second post.  I am doing that, but I want to see what you are seeing.  So, tell me, in your second post, what MavLink message did it fail on?

Was it attitude?  Was it batterystatus?  What was the message that it failed on?  If you tell me that then I will understand, but I don't see it yet, so please tell me what message it failed on.

kgmormylo, 2018-11-01:
Enlade,
That's three diffent log file after three startup programm.
Look it closely

io.dronefleet.mavlink.common.ActuatorControlTarget this string is type of message and error occurs on different message type. Only than APM send me V2 messages. Than it send V1 messages all work fine.

[byte_log.txt](https://github.com/dronefleet/mavlink/files/2538612/byte_log.txt)
[byte_log.txt](https://github.com/dronefleet/mavlink/files/2538616/byte_log.txt)
[byte_log.txt](https://github.com/dronefleet/mavlink/files/2538617/byte_log.txt)

Enlade, 2018-11-01:
Here is an excerpt from one of your posts above.  It shows what you print before it fails:

```
Message type simple name ServoOutputRaw
MavlinkMessage{originSystemId=1, originComponentId=1, payload=ServoOutputRaw{timeUsec=2742753101, port=0, servo1Raw=900, servo2Raw=900, servo3Raw=900, servo4Raw=900, servo5Raw=900, servo6Raw=900, servo7Raw=900, servo8Raw=900, servo9Raw=0, servo10Raw=0, servo11Raw=0, servo12Raw=0, servo13Raw=0, servo14Raw=0, servo15Raw=0, servo16Raw=0}}
fromV2
MavlinkPacket constructor
deserialize
Exception in thread "main" java.lang.ArrayIndexOutOfBoundsException
	at java.lang.System.arraycopy(Native Method)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.lambda$5(ReflectionPayloadDeserializer.java:80)

```

See, you print out ServoOutputRaw and then after that you print out the entire ServoOutputRaw message with all of its fields.  If you can print out all of its fields then that message has already been deserailized.  So that is NOT the message that the code is failing on.

After you print out that message you see the call to a new MAVLink constructor.  That means you are creating a new message that you want to fill up from the stream by deserializing the payload.  I want to know what that new message is because that is the message that is failing.  It is NOT failing on the ServoOutputRaw message.  It is failing on some other message and your posts don't tell us what it is.  Right?

kgmormylo, 2018-11-01:
you rigth. ServoOutputRaw is last correct message.

Enlade, 2018-11-01:
That's not a for instance.  That's a hey I'll tell you the message ID.  That is what I am asking you to do.

Enlade, 2018-11-01:
None of your examples prior told me that information.  The message it failed on was the VfrHud message.  Now, do that and add back in the code that tells you the offset/lengths and maybe print the Hex string payload as well.  Then we can begin.  :)

kgmormylo, 2018-11-01:
ok. and that example show that i try to tell you.
it doens't depend of message type.

```
Start
...
MessageName io.dronefleet.mavlink.common.AttitudeQuaternion
arraycopy occured
MessageName io.dronefleet.mavlink.common.AttitudeQuaternion
arraycopy occured
MessageName io.dronefleet.mavlink.common.AttitudeQuaternion
arraycopy occured
MessageName io.dronefleet.mavlink.common.AttitudeQuaternion
arraycopy occured
MessageName io.dronefleet.mavlink.common.AttitudeQuaternion
arraycopy occured
MessageName io.dronefleet.mavlink.common.AttitudeQuaternion
arraycopy occured
MessageName io.dronefleet.mavlink.common.AttitudeQuaternion
arraycopy occured
MessageName io.dronefleet.mavlink.common.AttitudeQuaternion
arraycopy occured
Message type simple name AttitudeQuaternion
MavlinkMessage{originSystemId=1, originComponentId=1, payload=AttitudeQuaternion{timeBootMs=1394338, q1=0.14851694, q2=0.028752154, q3=-0.015878728, q4=-0.9883643, rollspeed=-1.64086E-4, pitchspeed=9.823404E-4, yawspeed=-6.9655105E-4}}
fromV2
MavlinkPacket constructor
deserialize
MessageName io.dronefleet.mavlink.common.LocalPositionNed
arraycopy occured
MessageName io.dronefleet.mavlink.common.LocalPositionNed
arraycopy occured
MessageName io.dronefleet.mavlink.common.LocalPositionNed
arraycopy occured
MessageName io.dronefleet.mavlink.common.LocalPositionNed
arraycopy occured
MessageName io.dronefleet.mavlink.common.LocalPositionNed
arraycopy occured
MessageName io.dronefleet.mavlink.common.LocalPositionNed
arraycopy occured
MessageName io.dronefleet.mavlink.common.LocalPositionNed
arraycopy occured
Message type simple name LocalPositionNed
MavlinkMessage{originSystemId=1, originComponentId=1, payload=LocalPositionNed{timeBootMs=1394338, x=0.0, y=0.0, z=-0.6047778, vx=0.013350165, vy=-0.0033859068, vz=-0.61485726}}
fromV2
MavlinkPacket constructor
fromV2
MavlinkPacket constructor
deserialize
MessageName io.dronefleet.mavlink.common.BatteryStatus
arraycopy occured
MessageName io.dronefleet.mavlink.common.BatteryStatus
arraycopy occured
MessageName io.dronefleet.mavlink.common.BatteryStatus
arraycopy occured
MessageName io.dronefleet.mavlink.common.BatteryStatus
arraycopy occured
MessageName io.dronefleet.mavlink.common.BatteryStatus
arraycopy occured
MessageName io.dronefleet.mavlink.common.BatteryStatus
arraycopy occured
MessageName io.dronefleet.mavlink.common.BatteryStatus
arraycopy occured
MessageName io.dronefleet.mavlink.common.BatteryStatus
arraycopy occured
MessageName io.dronefleet.mavlink.common.BatteryStatus
arraycopy occured
Message type simple name BatteryStatus
MavlinkMessage{originSystemId=1, originComponentId=1, payload=BatteryStatus{id=0, batteryFunction=EnumValue{value=1, entry=MAV_BATTERY_FUNCTION_ALL}, type=EnumValue{value=1, entry=MAV_BATTERY_TYPE_LIPO}, temperature=32767, voltages=null, currentBattery=-1, currentConsumed=-1, energyConsumed=-1, batteryRemaining=-1, timeRemaining=0, chargeState=null}}
fromV2
MavlinkPacket constructor
deserialize
MessageName io.dronefleet.mavlink.common.Timesync
arraycopy occured
MessageName io.dronefleet.mavlink.common.Timesync
Exception in thread "main" java.lang.ArrayIndexOutOfBoundsException
	at java.lang.System.arraycopy(Native Method)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.lambda$5(ReflectionPayloadDeserializer.java:85)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(Unknown Source)
	at java.util.stream.ReferencePipeline$2$1.accept(Unknown Source)
	at java.util.ArrayList.forEach(Unknown Source)
	at java.util.stream.SortedOps$RefSortingSink.end(Unknown Source)
	at java.util.stream.Sink$ChainedReference.end(Unknown Source)
	at java.util.stream.AbstractPipeline.copyInto(Unknown Source)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(Unknown Source)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(Unknown Source)
	at java.util.stream.AbstractPipeline.evaluate(Unknown Source)
	at java.util.stream.ReferencePipeline.forEach(Unknown Source)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.deserialize(ReflectionPayloadDeserializer.java:68)
	at io.dronefleet.mavlink.MavlinkConnection.next(MavlinkConnection.java:273)
	at simpleMavlink.Main.main(Main.java:111)
```

kgmormylo, 2018-11-01:
> None of your examples prior told me that information. The message it failed on was the VfrHud message. Now, do that and add back in the code that tells you the offset/lengths and maybe print the Hex string payload as well. Then we can begin. :)

Ok. How to add hex string? Do you want hex string of payload?

Enlade, 2018-11-01:
It might be helpful to dump the payload with that little trick I did to give the hex string instead of all the nonprintable characters, something like:

`System.out.println(String.format("%X", new BigInteger(1, payload)));
`

But also toss in the offset/lengths as well.

kgmormylo, 2018-11-01:
```
deserialize
MessageName io.dronefleet.mavlink.common.EstimatorStatus
Payload = AA7CF961000000000000000078B0B438A5416A3DD8E0293C00000000000000008C31F43DB137753EA5
offset = 0length = 8
arraycopy occured
offset = 8length = 4
arraycopy occured
offset = 12length = 4
arraycopy occured
offset = 16length = 4
arraycopy occured
offset = 20length = 4
arraycopy occured
offset = 24length = 4
arraycopy occured
offset = 28length = 4
arraycopy occured
offset = 32length = 4
arraycopy occured
offset = 36length = 4
arraycopy occured
offset = 40length = 2
Exception in thread "main" java.lang.ArrayIndexOutOfBoundsException
	at java.lang.System.arraycopy(Native Method)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.lambda$5(ReflectionPayloadDeserializer.java:90)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(Unknown Source)
	at java.util.stream.ReferencePipeline$2$1.accept(Unknown Source)
	at java.util.ArrayList.forEach(Unknown Source)
	at java.util.stream.SortedOps$RefSortingSink.end(Unknown Source)
	at java.util.stream.Sink$ChainedReference.end(Unknown Source)
	at java.util.stream.AbstractPipeline.copyInto(Unknown Source)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(Unknown Source)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(Unknown Source)
	at java.util.stream.AbstractPipeline.evaluate(Unknown Source)
	at java.util.stream.ReferencePipeline.forEach(Unknown Source)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.deserialize(ReflectionPayloadDeserializer.java:73)
	at io.dronefleet.mavlink.MavlinkConnection.next(MavlinkConnection.java:273)
	at simpleMavlink.Main.main(Main.java:111)
```

Enlade, 2018-11-01:
Interesting...are you using the new comparator code that ben posted?  It seems like the field order is off a bit.  I think the order should be [8, 2, 4, 4, 4, 4, 4, 4, 4, 4] instead of [8, 4, 4, 4, 4, 4, 4, 4, 4, 2].  Now, the payload also seems a byte short, but that could be because of the trick that I did.  Any leading 0's will be dropped.  So, it is possible that the payload started with 00 which might give you the right size.  Maybe you could also print out the payload length when you print it.  You could modify my little one line HEX dump to left pad the hex string with 0's up to the length of the payload and that way it would avoid this problem, I was being lazy.  In any case, if you print out the length of the payload then we will know if we need to add 0's to the left side or we will know if the payload is actual a byte short.

kgmormylo, 2018-11-01:
I use last version 1.0.12 of posted code.
If i understood you rignth

```
			System.out.println(String.format("Payload = " + "%X", new BigInteger(1, payload)));
		    System.out.println("payload length " + payload.length);
```
```
fromV2
MavlinkPacket constructor
deserialize
MessageName io.dronefleet.mavlink.common.Ping
Payload = BF8A538E000000004D09
payload length 10
offset = 0 length = 8
arraycopy occured
offset = 8 length = 4
Exception in thread "main" java.lang.ArrayIndexOutOfBoundsException
	at java.lang.System.arraycopy(Native Method)
```

Enlade, 2018-11-01:
Use this to do the hex dump instead.  There are leading 0's data that that little code loses, but this should show it.

`    	System.out.println(String.format("%" + (payload.length*2) + "s", String.format("%X", new BigInteger(1, payload))).replace(" ", "0"));
`

kgmormylo, 2018-11-01:
```
fromV2
MavlinkPacket constructor
deserialize
MessageName io.dronefleet.mavlink.common.EstimatorStatus
4A9C66FD0000000000000000FA8DB6381081723DD42F003C0000000000000000EEE0F53DA8937A3EA5
offset = 0 length = 8
arraycopy occured
offset = 8 length = 4
arraycopy occured
offset = 12 length = 4
arraycopy occured
offset = 16 length = 4
arraycopy occured
offset = 20 length = 4
arraycopy occured
offset = 24 length = 4
arraycopy occured
offset = 28 length = 4
arraycopy occured
offset = 32 length = 4
arraycopy occured
offset = 36 length = 4
arraycopy occured
offset = 40 length = 2
Exception in thread "main" java.lang.ArrayIndexOutOfBoundsException
	at java.lang.System.arraycopy(Native Method)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.lambda$5(ReflectionPayloadDeserializer.java:89)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(Unknown Source)
	at java.util.stream.ReferencePipeline$2$1.accept(Unknown Source)
	at java.util.ArrayList.forEach(Unknown Source)
	at java.util.stream.SortedOps$RefSortingSink.end(Unknown Source)
	at java.util.stream.Sink$ChainedReference.end(Unknown Source)
	at java.util.stream.AbstractPipeline.copyInto(Unknown Source)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(Unknown Source)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(Unknown Source)
	at java.util.stream.AbstractPipeline.evaluate(Unknown Source)
	at java.util.stream.ReferencePipeline.forEach(Unknown Source)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.deserialize(ReflectionPayloadDeserializer.java:72)
	at io.dronefleet.mavlink.MavlinkConnection.next(MavlinkConnection.java:273)
	at simpleMavlink.Main.main(Main.java:111)
```

Enlade, 2018-11-01:
Yea, that has length of 10 when the ping should have length of 14.  Means the problem isn't in the deserializer, but rather in the code that builds the payload from the stream.  Mind you, there may also be a problem with the way the comparator is working because the order of that EstimatorStatus message was a bit off (or it seemed as such).  However, the problem is that the payload is not complete for some reason.  That is likely in  MavlinkPacket.java source code or there abouts.

kgmormylo, 2018-11-01:
sorry, but i couldn't understand your explanation clearly. How did you determine that current length is 10.

Enlade, 2018-11-01:
I was talking about your previous post where you printed the payload.length for the ping message.  The number of bytes that you printed for the payload was 10 and the payload.length you printed was 10.  However, a ping message requires 14 bytes for the payload.  So, the length of the payload is off.  That means that the problem is somewhere else in the code.  Likely it is in MavlinkPacket.java or it is in MavlinkPacketReader.java.

Your second post has 41 bytes (though you didn't print payload length there).  It should be 42 bytes.  But I'm not sure if there is or isn't a leading 00 without seeing the payload length printed (unless maybe you have added in my new little hex dump code that pads the 0's?).  Since the error occurs then it is likely that it is one byte short as well.  This again suggests that the reader part of the code is where the problem is, and not in the deserialize code (where it crashes).

Mind you, and I want to point this out again because it is some other issue that should be investigated.  The EstimatorStatus message should be deserialized as [8, 2, 4, 4, 4, 4, 4, 4, 4, 4], but your code is suggesting that it is trying to do it as [8, 4, 4, 4, 4, 4, 4, 4, 4, 2].  This could suggest that there is another problem with the comparator that is waiting to bite you.  But you can ignore that for now because your bigger problem is that the reader is not giving you the right payload length.

I have reviewed the code there and it all seems good, so this is particularly sticky.  Still, I don't see any obvious code bugs in either of those .java files.  I'll keep looking.

Enlade, 2018-11-01:
Maybe it has something to do with ByteArray.java and how it manages any exceptions that it might need to handle.  I'm not sure off hand how copyarray (for slice) or such similar byte array methods will deal with position and size parameters that are out of bounds.  Some languages will simply give you as much as they can.  For instance, if you want to grab 10 characters out of some string starting at position 10, but if the string is only 15 characters long, sometimes languages will still return 5 characters (to end of string).  My guess is that JAVA pops out an exception instead, but if not then maybe your input stream is noisy and you are missing some data and these functions slice it up best they can to give you something.  Of course that something just crashes the program later during the deserializing.  I really don't think that is the case though, but I'm pulling at straws now, so just tossing it out there.

benbarkay, 2018-11-01:
Can a recording of packets that reproduce this be provided? Otherwise I will have to set up a simulation to reproduce this failure and that might take me a while to get to given how busy things are at the moment.

Enlade, 2018-11-01:
Hey, you should add this code to the MavlinkConnection.java file before the call to the deserializer.

`System.out.println("===================================" + String.format("%" + (packet.getRawBytes().length*2) + "s", String.format("%X", new BigInteger(1, packet.getRawBytes()))).replace(" ", "0"));
`

This will dump the entire raw message for us so that we can see it before the failure.  I am interested in whether the message length is reported as 14 bytes in the raw message by that the payload itself is only 10 bytes (or whatever it is for whatever message, that was for the ping).  So, lets get a look at the actual raw message data on which it is failing.  Give it a try.

Enlade, 2018-11-01:
It is magic byte followed by payload length byte...so the second byte is what we are after, and you want to print out the payload.length as well (the byte[] length), like you did before.

Enlade, 2018-11-01:
Or do this instead cause it looks much prettier:

`System.out.println("================" + String.format("%" + (packet.getRawBytes().length*2) + "s", String.format("%X", new BigInteger(1, packet.getRawBytes()))).replace(" ", "0").replaceAll("..", "$0 "));`

kgmormylo, 2018-11-02:
> Can a recording of packets that reproduce this be provided? Otherwise I will have to set up a simulation to reproduce this failure and that might take me a while to get to given how busy things are at the moment.

Ben, sorry. I am newby in Java. Which form of packet i should record?

kgmormylo, 2018-11-02:
Enlade, i add this code to method fromV2Bytes of class MavlinkPacket

```
    public static MavlinkPacket fromV2Bytes(byte[] rawBytes) {
    	System.out.println("fromV2");
        ByteArray bytes = new ByteArray(rawBytes);
        System.out.println("rawBytes" + String.format("%" + (rawBytes.length*2) + "s", String.format("%X", new BigInteger(1, rawBytes))).replace(" ", "0").replaceAll("..", "$0 "));
        int versionMarker = bytes.getInt8(0);
        System.out.println("versionMarker = " + versionMarker);
        int payloadLength = bytes.getInt8(1);
        System.out.println("payloadLength = " + payloadLength);
        int incompatibleFlags = bytes.getInt8(2);
        System.out.println("imcompatibleFlags = " + incompatibleFlags);
        int compatibleFlags = bytes.getInt8(3);
        System.out.println("compatibleFlags = " + compatibleFlags);
        int sequence = bytes.getInt8(4);
        System.out.println(" sequence = " +  sequence );
        int systemId = bytes.getInt8(5);
        int componentId = bytes.getInt8(6);
        int messageId = bytes.getInt24(7);
        System.out.println(" messageId = " + messageId);
        byte[] payload = bytes.slice(10, payloadLength);
        int checksum = bytes.getInt16(10 + payloadLength);
        byte[] signature = bytes.slice(12 + payloadLength, 13);
```
Result of programm

```
fromV2
rawBytes = FD 11 00 00 6E 01 01 4A 00 00 00 00 00 00 1D 49 5B 3D 4B E6 5F 40 6E B1 81 BE 69 EE 56 FD 1C 00 00 6F 01 01 1E 00 00 8C 48 0A 00 56 
versionMarker = 253
payloadLength = 17
imcompatibleFlags = 0
compatibleFlags = 0
 sequence = 110
 messageId = 74
MavlinkPacket constructor
deserialize
MessageName io.dronefleet.mavlink.common.VfrHud
000000001D495B3D4BE65F406EB181BE69
payload length = 17
offset = 0 length = 4
arraycopy occured
offset = 4 length = 4
arraycopy occured
offset = 8 length = 4
arraycopy occured
offset = 12 length = 4
arraycopy occured
offset = 16 length = 2
Exception in thread "main" java.lang.ArrayIndexOutOfBoundsException
```

benbarkay, 2018-11-02:
This is a VRF_HUD message, it should have a payload length of 18, but was
truncated to 17 as specified here:
https://mavlink.io/en/guide/serialization.html#payload_truncation
(Empty-Byte Payload Truncation (MAVLink 2))

Opened issue #17
Thank you @kgmormylo

kgmormylo, 2018-11-02:
Thanks Ben and Enlade for help understanding. I close this issue and waiting update.

closed 2018-11-02 by kgmormylo