dronefleet/mavlink #16: MavlinkMessage to byte array

gshmakov, 2018-11-01:
Hi,
Is there any way to convert MavlinkMessage into byte array (in another words get raw byte representation of single mavlink message)?

Enlade, 2018-11-01:
Here is a quick way (thought not efficient) to convert a byte[] into a HEX string and print it to system out.

System.out.println(String.format("%" + (payload.length*2) + "s", String.format("%X", new BigInteger(1, payload))).replace(" ", "0"));

In this case I am printing the payload portion of the message, but ignore that and just assume payload is any arbitrary byte[].

So, you just need to figure out where in the code you have a possible packet and insert that to print out the entire packet.

I think maybe you can do what you are after by printing the frame that is used to generate the V1 or V2 MavlinkMessage.  So, inside of MavlinkPacketReader.java you will see a case statement.  After a frame is determined to be a V1 or V2 message you could toss in a hex print like I have suggested above.  You don't want to do it for every frame, but only those frames that are possible V1 and V2 messages.  Of course, you will likely get extra frames printing in this case, but you should get the frames coming in.

Enlade, 2018-11-01:
Now, if you wanted to be more fancy you could add a method to MavLinkPacket and call it hexToString or some such.  That method could push everything back into a single hex string for you and spit it back out when you want to print it as such.  Again, that would not be a hard change.  I just was not sure if you wanted this for debugging or if you wanted it for some feature of your program.  If it is to be a feature then tossing in a toHexString method would probably be best.

Enlade, 2018-11-01:
Oh wait, there is a rawbytes method in MavLinkPacket already.  Probably you could just grab that and use my little method above to print it.  Though we are starting to stress my little hex dump trick I think.  Might need to create a proper hex dump method instead.

Enlade, 2018-11-01:
Oh, lets make it look prettier by putting in spaces between hex bytes, use this instead:

`System.out.println("================" + String.format("%" + (packet.getRawBytes().length*2) + "s", String.format("%X", new BigInteger(1, packet.getRawBytes()))).replace(" ", "0").replaceAll("..", "$0 "));
`

Enlade, 2018-11-01:
Ahhhh, yes, I see what you are saying ben, the packet is not made available to the MavlinkMessage.  So, it is not available for printing within the applications message handling loop.

He would have to add the rawbytes to MavlinkMessage and pass it up on creation in MavlinkConnection for it to be accessible to the application (getting messy).  I can't really see a reason to want to do that though, other than for debugging, and he can add the above line to MavlinkConnection to do debugging on the raw message.  The application has access to the payload to print if it wanted, and that's probably more then needed to print at that level.  The fields of the messages is better to print and you have a toString for that in each message (and we have both posted different ways to get that into different formats as needed).

So, I am just assuming that the raw bytes is only needed for debugging, the above suggestion should do that.

benbarkay, 2018-11-02:
@gshmakov you can use this class to create an `InputStream` from a `DatagramSocket`:

```java
import java.io.IOException;
import java.io.InputStream;
import java.net.DatagramPacket;
import java.net.DatagramSocket;

public class UdpInputStream extends InputStream {
    private final DatagramSocket socket;
    private final byte[] buffer;
    private int packetLength;
    private int cursor;

    public UdpInputStream(DatagramSocket socket, byte[] buffer) {
        this.socket = socket;
        this.buffer = buffer;
    }

    @Override
    public int read() throws IOException {
        if (cursor < packetLength) {
            return buffer[cursor++] & 0xff;
        } else {
            readNext();
            return read();
        }
    }

    private void readNext() throws IOException {
        DatagramPacket packet = new DatagramPacket(buffer, buffer.length);
        socket.receive(packet);
        packetLength = packet.getLength();
        cursor = 0;
    }
}
```

Then use it like so:
```java
UdpInputStream udpIn = new UdpInputStream(myUdpSocket);
MavlinkFrameReader frameReader = new MavlinkFrameReader(udpIn);
while (frameReader.next()) {
    byte[] frameBytes = frameReader.frame();
    // you can use frameReader.drop() here if you wish to drop the packet, if
    // it seems corrupted or not understood.
}
```

benbarkay, 2018-11-02:
You also don't need to depend on `mavlink`, you can just depend on `mavlink-protocol`:
https://github.com/dronefleet/mavlink/tree/master/mavlink-protocol

Enlade, 2018-11-02:
So, once he has the frameBytes can he use the deserializer just to confirm that it is a valid MAVLink message before he sends it out to Mission Planner?  That is, go ahead and generate the Mavlink package and deserialize the payload, and if that all works, then package the frameBytes up and send it to Mission Planner.  That way he is only sending out valid Mavlink messages to Mission Planner.

gshmakov, 2018-11-02:
@benbarkay @Enlade thank you for your help. 
Problem solved by use of MavlinkFrameReader instead of MavlinkConnection.

closed 2018-11-02 by gshmakov