dronefleet/mavlink #17: Handle payload truncation

benbarkay, 2018-11-02:
Current implementation does not support Mavlink2 "Empty-Byte Payload Truncation" as specified here:
https://mavlink.io/en/guide/serialization.html#payload_truncation

Enlade, 2018-11-02:
Just to note one other thing from kgmormylo's data that might be a different issue that might need fixing.  I could be wrong about my understanding of this, but I want to just mention it just in case.  I'll just point out two examples from his code.

1.  For his VrfHud message he shows an ordering for the field lengths of [4, 4, 4, 4, 2] before it crashes.  However, I think that it should have been [4, 4, 2, 2, 4, 4].

2.  For his EstimatorStatus message he shows an ordering for the field lengths of [8, 4, 4, 4, 4, 4, 4, 4, 4, 2].  However, I think that it should have benn [8, 2, 4, 4, 4, 4, 4, 4, 4, 4].

Just something to consider while you're working on the V2 fixes.

benbarkay, 2018-11-02:
@Enlade the protocol packs the fields in the way specified by `WIRE_COMPARATOR`, that is from longer to shorter bit length, this part seems to be done correctly.

closed 2018-11-03 by benbarkay

kgmormylo, 2018-11-07:
Checked. Error disappeared.

cjrustic, 2019-05-06:
An error occurred while the rcv-packet was parsed, it does not seem to be converted in the order of its position.
In io.dronefleet.mavlink.util.WireFieldInfoComparator, I tried to delete the field length comparison, so I got the correct result. Is it correct?

benbarkay, 2019-05-08:
@rusy could you send me some bytes?
I have been on relatively long leave, so it may take some time for me to
set things up to make a release, but having this ready will surely speed
things up.

Thank you for reporting the issue :)

cjrustic, 2019-05-09:
Thanks for your reply.
In view of this issue was closed, then I deleted my previous message.

I have a quad-drone using Pixhawk(2.4.8? RadioLink), and a Pixhawk4(HolyBro) with GPS module only. Obtain GPS information continuously is my first target now, so I started from Heartbeat.
However, I thought it is a problem while received all packets, but actually it's not. I have checked the others messages, everything was fine. Now I still confuse the process about parse a packet using its length. 

I would greatly appreciate it if you kindly give me some feedback on this ...

<original>
------------------------------------
[image]

<remark the length check in WireFieldInfoComparator>
------------------------------------
[image]

commit 5cd36218 (2018-11-03): fix: #17