dronefleet/mavlink #18: Error slicing byte array when parsing V1 packet

twsmith85, 2018-12-12:
I am running into an 'array out of bounds' error using the library with Android over a USB connection. I'm opening the connection as a serial port using https://github.com/felHR85/UsbSerial and then obtaining streams from that. The error I get back is below. Any ideas why I would get 0xFE's back to back? I'm digging further into the iostreams currently, but wondering if you have any ideas from your side. Thanks!

Exception thrown to log:
```
E/MAVLINk: Caught Array bounds exception reading mavlink messages.
    java.lang.ArrayIndexOutOfBoundsException: src.length=6 srcPos=6 dst.length=254 dstPos=0 length=254
        at java.lang.System.arraycopy(Native Method)
        at io.dronefleet.mavlink.protocol.ByteArray.slice(ByteArray.java:44)
        at io.dronefleet.mavlink.protocol.MavlinkPacket.fromV1Bytes(MavlinkPacket.java:130)
        at io.dronefleet.mavlink.protocol.MavlinkPacketReader.next(MavlinkPacketReader.java:39)
        at io.dronefleet.mavlink.MavlinkConnection.next(MavlinkConnection.java:213)
        at com.company.app.service.commProtocols.Mavlink.Mavlink$ReaderThread.run(Mavlink.java:247)
I/MAVLINk: Closing Mavlink connection
```

Debugger output:
```
INCOMPAT_FLAG_SIGNED = 1 (0x1)
 MAGIC_V1 = 254 (0xFE)
 MAGIC_V2 = 253 (0xFD)
 serialVersionUID = 3105533922443436538 (0x2B1912A60D2AA1FA)
rawBytes = {byte[6]@6682} 
 0 = -2 (0xFE)
 1 = -2 (0xFE)
 2 = 9 (0x9)
 3 = 51 (0x33)
 4 = 68 (0x44)
 5 = 0 (0x0)
bytes = {ByteArray@6683} 
 bytes = {byte[6]@6682} 
 shadow$_klass_ = {Class@6605} "class io.dronefleet.mavlink.protocol.ByteArray"
 shadow$_monitor_ = -2076735974 (0x8437861A)
versionMarker = 254 (0xFE)
payloadLength = 254 (0xFE)
```

benbarkay, 2018-12-13:
Which version are you using? Is it the newest?

benbarkay, 2018-12-13:
I am not able to reproduce this locally, I've tried:
```java
    @Test
    public void issue18_mavlink1ArrayIndexOutOfBounds() throws IOException {
        byte[] packetByes = {(byte)0xFE, (byte)0xFE, 0x9, 0x33, 0x44, 0};
        packetByes = Arrays.copyOf(packetByes, 1024);

        InputStream in = new ByteArrayInputStream(packetByes);
        OutputStream out = new ByteArrayOutputStream();

        MavlinkConnection connection = MavlinkConnection.create(in, out);
        connection.next();
    }
```
What happens is that a packet of the right size is read, so no exception is thrown and the issue is not reproduced.

What is interesting to find out is how it happened that `MavlinkPacket#fromV1Bytes` received a byte array of 6 when the payload length denotes that the frame is 260 bytes (header + payload). `MavlinkFrameReader` shouldn't permit this, and I can't manage to make this happen using `ByteArrayInputStream` -- This is likely something to do with either the USB library's implementation of `InputStream` or some Android discrepancy from Java 8.

twsmith85, 2018-12-13:
I am running the newest version from master. I agree with you on the frame reader. I don't understand how I get to this point with the checks I see in the code.

I'm guessing it might have something to do with pushing bytes back to the stream. We are using Java 8 in android for our build, so it should be fairly close in that regard.

I am planning to obtain the iostreams a different way today and see if that solves the problem.

Also, this is on the PX4 from 3DR just as an added note.

Thanks again for the feedback!

benbarkay, 2018-12-13:
I'll spend some more time trying to reproduce this on Saturday. It's surely possible to catch this earlier (`TransactionalInputStream` failing to advance somehow). While it's possible that there is some implementation failure with the InputStream from the USB library (seems like they recently changed code there in Master), I'd still not like dronefleet/mavlink to fail in this fashion.

I'm curious, does this happen when the USB connection closes, or intermittently?

benbarkay, 2018-12-14:
I tried reading from an InputStream which generates random bytes for about an hour but couldn't reproduce this.

I've just released 1.1.3 which fixes a seemingly unrelated issue. I doubt it, but maybe this version will also solve your issue.

I suppose that the next step would be to on Android. Please let me know if you have any progress in finding out the cause of this :)

twsmith85, 2018-12-15:
Hey Ben, got backed up on testing this piece. I'll try to figure it out or close this ticket one way on Monday. I think I'm agreeing with you that it's most likely a stream or buffer failure.

twsmith85, 2018-12-18:
Sorry for the delay, I'm still debugging through this on Android. Slowing down the processing with added output seems to have hidden the original buffer overflow problem (Most likely because I'm not processing enough serial input to hit it yet). The serial reads seem to be working fine, I'm still digging through the 5 flavors of extended input stream to understand what I'm seeing.

Here's a couple of notes on what I'm seeing currently:

**1**. I found one minor bug that prevents the occasional packet from being identified on Android for me, not sure if this is the case in pure Java. In `MavlinkFrameReader.java -> next()`, Occasionally the read returns 0xF* bytes in padded integer form as 0xFFFFFFF* for me and this does not match the version check. The read call returns this padded integer very sporadically, I'm not sure why it doesn't happen every time. But it typically happens on the first packet for me at least.

```	12-18 10:37:50.751 29166 29234 I System.out: SER read: FE 09 86 01 01 00 00 00 00 00 0D 03 51 03 03 42 E4 
	12-18 10:37:50.751 29166 29234 I System.out: SER running byte count: 17
	12-18 10:37:50.752 29166 29234 I System.out: FR - Testing FFFFFFFE for version marker and 09 for payload length
	12-18 10:37:50.754 29166 29234 I System.out: FR header not found, rollback and skip```

 I would propose using the same '& 0xff' on these reads that is used in `MavlinkPacketReader.java -> next()`.

**2**. Every frame read in `MavlinkFrameReader.java -> next()` seems to have 2 extra bytes on the end and loses the sequence byte and first checksum byte. In this log, serial packet `FE 09 DC 01 01 00 00 00 00 00 0D 03 51 03 03 4D 63` becomes `FE 09 01 01 00 00 00 00 00 0D 03 51 03 03 63 FE 09 ` somehow by the time it gets back up to packet reader. I'm guessing this is partly due to `MavlinkFrameReader -> next()` reading versionMarker and payloadLength, then reading 6 or 11 more header bytes in the advance call. I'm guessing the advance call should use 4 and 9 as the hard coded additional header lengths? I'm not sure why its the sequence and checksum[0] bytes i'm losing yet.

```PR = MavlinkPacketReader
FR = MavlinkFrameReader
TIS = TransactionalInputStream
SER = SerialPortRead

12-18 11:23:57.324   308   375 I MY-MAVLINk: Opening serial port
12-18 11:23:57.362   308   375 I MY-MAVLINk: Serial port open
12-18 11:23:57.362   308   375 I MY-MAVLINk: Opening Mavlink
12-18 11:23:57.387   308   375 I MY-MAVLINk: Mavlink connection active
12-18 11:23:57.387   308   376 I MY-MAVLINk: Mavlink reader thread started.
12-18 11:23:57.387   308   376 I MY-MAVLINk: Requesting mavlink message.
12-18 11:23:57.387   308   376 I System.out:     TIS commit, cbuffer: 0
12-18 11:23:57.392   308   376 I System.out: SER read: FE 09 DA 01 01 00 00 00 00 00 0D 03 51 03 03 20 6D 
12-18 11:23:57.392   308   376 I System.out: SER running byte count: 17
12-18 11:23:57.393   308   376 I System.out: FR - Testing FE for version marker and 09 for payload length
12-18 11:23:57.393   308   376 I System.out: FR V1 header found, payload len: 9
12-18 11:23:57.396   308   375 I MY-MAVLINk: TX: Heartbeat - 255:0 - Heartbeat{type=EnumValue{value=6, entry=MAV_TYPE_GCS}, autopilot=EnumValue{value=8, entry=MAV_AUTOPILOT_INVALID}, baseMode=null, customMode=0, systemStatus=EnumValue{value=0, entry=MAV_STATE_UNINIT}, mavlinkVersion=3}
12-18 11:23:57.397   308   376 I System.out: SER read: FE 09 DB 01 01 00 00 00 00 00 0D 03 51 03 03 CA 13 
12-18 11:23:57.397   308   376 I System.out: SER running byte count: 35
12-18 11:23:57.402   308   376 I System.out: SER read: FE 09 DC 01 01 00 00 00 00 00 0D 03 51 03 03 4D 63 
12-18 11:23:57.402   308   376 I System.out: SER running byte count: 53
12-18 11:23:57.407   308   376 I System.out: SER read: FE 09 DD 01 01 00 00 00 00 00 0D 03 51 03 03 A7 1D 
12-18 11:23:57.407   308   376 I System.out: SER running byte count: 71
12-18 11:23:57.407   308   376 I System.out: PR get a frame
12-18 11:23:57.407   308   376 I System.out: PR frame running byte count: 17
12-18 11:23:57.412   308   376 I System.out: PR frame read: FE 09 01 01 00 00 00 00 00 0D 03 51 03 03 63 FE 09 
12-18 11:23:57.412   308   376 I System.out: PR found V1 packet
12-18 11:23:57.415   308   376 I System.out:     TIS Rolled back 17 bytes (FE 09 01 01 00 00 00 00 00 0D 03 51 03 03 63 FE 09) to buffer, cbuffer: 17
12-18 11:23:57.415   308   376 I System.out:     TIS commit, cbuffer: 0
12-18 11:23:57.416   308   376 I System.out:     TIS commit, cbuffer: 0
12-18 11:23:57.416   308   376 I System.out: FR - Testing 09 for version marker and 01 for payload length
12-18 11:23:57.416   308   376 I System.out: FR header not found, rollback and skip
12-18 11:23:57.416   308   376 I System.out:     TIS Rolled back 2 bytes (09 01) to buffer, cbuffer: 2
12-18 11:23:57.416   308   376 I System.out:     TIS commit, cbuffer: 0
12-18 11:23:57.417   308   376 I System.out: FR - Testing 01 for version marker and 01 for payload length
12-18 11:23:57.417   308   376 I System.out: FR header not found, rollback and skip
12-18 11:23:57.417   308   376 I System.out:     TIS Rolled back 2 bytes (01 01) to buffer, cbuffer: 2
12-18 11:23:57.417   308   376 I System.out:     TIS commit, cbuffer: 0
12-18 11:23:57.418   308   376 I System.out: FR - Testing 01 for version marker and 00 for payload length
12-18 11:23:57.418   308   376 I System.out: FR header not found, rollback and skip
[... cut ...]
12-18 11:23:57.444   308   376 I System.out:     TIS commit, cbuffer: 0
12-18 11:23:57.444   308   376 I System.out: FR - Testing 09 for version marker and 01 for payload length
12-18 11:23:57.444   308   376 I System.out: FR header not found, rollback and skip
```
I'm making these changes locally and continuing testing in search of the buffer overflows. If these changes make sense and you would rather have a pull request for them, just shout!

Thanks again for the open source code, the back and forth dialog, and testing!

-Tyler

twsmith85, 2018-12-18:
Looks like problem #1 may also be a failing of the InputStream class that I am using.

benbarkay, 2018-12-18:
As for 1, consider the following:
```java
int i = 0xFE; // = 254 = 11111110
byte b = (byte)i; // = -2 = 11111110
int c = b; // = -2 = 11111111111111111111111111111110
```

This is exactly what is happening here:
https://github.com/felHR85/UsbSerial/blob/master/usbserial/src/main/java/com/felhr/usbserial/SerialInputStream.java#L45

It's hard to support this because a value of `255` (i.e. 11111111) would be `-1` -- which is an EOF result. Do you want to make a pull-request to UsbSerial? :)

benbarkay, 2018-12-18:
Hopefully didn't come off as passive-aggressive there now that I read it again, contributions to `dronefleet/mavlink` would be very appreciated. It's just not as straightforward to support this sort of bit truncation from `InputStream#read()` as it may seem at first sight.

twsmith85, 2018-12-18:
Not at all. You're exactly right on. That was my conclusion and the cause of my last comment. I should have elaborated further.

I'm digging into the UsbSerial a bit further to see if there are any other bits causing me heartburn. After that, I'll most definitely send an issue and/or pull request their way.

Thanks much for your time in looking at this. 👍 

Closing this issue as it is solved by fixes to the SerialInputStream of the serial library.

closed 2018-12-18 by twsmith85