dronefleet/mavlink #19: LogData payload has null data array but the bytes are received in the message.

tysm-msab, 2019-01-23:
I am trying to pull logs from a pixhawk1 using the library, but I cannot seem to figure out where I am losing the data array. Example below show's what I'm seeing with 2 back to back calls in the reader thread. 

LogRequestData was set to match the 00 b4 00 00 that I sniffed as the first request from QGroundControl app which is working. I originally assumed this was a value of 180. It seems to be 46080 in the builder here.

**Request**
```
LogRequestData.builder()
                .id(mSource.id())
                .count(46080)
                .ofs(46080)
                .targetSystem(1)
                .targetComponent(1)
                .build();
```
**Response message.getPayload().toString():**
`RX v1:  LogData{id=2, ofs=46080, count=90, data=null}`
**Response byteArrayToHexString(message.getRawBytes()):**
`RX v1 Raw bytes: fe 61 5c 01 01 78 00 b4 00 00 02 00 5a af 37 58 92 f4 3d 7d 1c df bd 3b e9 19 c1 a3 95 87 f6 6d 00 00 00 08 f4 36 28 66 4d bb 4f d8 5e 3b 48 60 28 3e e0 ce 6c bd 75 d7 20 c1 a3 95 83 09 6e 00 00 c0 e5 46 3a 00 68 8e b7 dc a4 d0 b9 f8 05 fd 3d 08 48 b9 bd f6 d8 19 c1 a3 95 87 09 6e 00 00 dc 5b 40 bb 70 0a 2d 06 02`

From this, I can tell that most of the response seems to be parsed correctly:
- offset is the same as requested:  00 b4 00 00
- ID is the same as requested: 02 00
- byte count is the 90 requested: 5a

The response raw data is the same as what I saw over the wire previously. However, It seems that retrieving the payload as a LogData type does not reference or copy the array. I'm reading up on the deserializer, but haven't found the culprit yet.  Any helpful hints are greatly appreciated. Thanks!

benbarkay, 2019-01-24:
@tysm-msab 
I can already see the problem, there is no code to invoke the setter for `byte[]` in the deserializer. I'll make a fixed release this evening.

Thank you for opening the issue.

tysm-msab, 2019-01-24:
Awesome. I was just getting it integrated/building locally to test that myself. Thanks @benbarkay.

benbarkay, 2019-01-24:
@tysm-msab Let me know if this solves it, you may close this issue if it does

tysm-msab, 2019-01-24:
Seems to be working!

`RX V1: LogData{id=2, ofs=46080, count=90, data=[B@dc0b8e7} - fe 61 9d 01 01 78 00 b4 00 00 02 00 5a af 37 58 92 f4 3d 7d 1c df bd 3b e9 19 c1 a3 95 87 f6 6d 00 00 00 08 f4 36 28 66 4d bb 4f d8 5e 3b 48 60 28 3e e0 ce 6c bd 75 d7 20 c1 a3 95 83 09 6e 00 00 c0 e5 46 3a 00 68 8e b7 dc a4 d0 b9 f8 05 fd 3d 08 48 b9 bd f6 d8 19 c1 a3 95 87 09 6e 00 00 dc 5b 40 bb 70 0a 2d b0 4f`

Thanks much @benbarkay !

closed 2019-01-24 by tysm-msab