dronefleet/mavlink #20: Custom Mode using MAV_CMD_DO_SET_MODE?

Enlade, 2019-01-24:
I am having trouble getting MAV_CMD_DO_SET_MODE to do anything, where using the deprecated Set_Mode everything works fine.

If I use Set_Mode with a BASE_MODE=89 and a CUSTOM_MODE=4 the UAV will respond in the heartbeat with a BASE_MODE=217 and a CUSTOM_MODE=4.  I have read that the BASE_MODE is ignored and only the CUSTOM_MODE matters, but I set the BASE_MODE anyways.  In any case, doing that with Set_Mode works fine and I can issue other commands to the UAV (like takeoff and such).

However, when I try the same thing using MAV_CMD_DO_SET_MODE I am not having much luck.  I set the param1=89 (BASE_MODE) and I set param2=4 (CUSTOM_MODE) and the responding Heartbeat changes mode to BASE_MODE=209 and CUSTOM_MODE=0.  I have tried setting the parameters to numerous other values and it always gives me 209 and 0 back. Although I can arm the UAV from this mode, I am not able to perform many of the other tasks I want to perform (like takeoff).  I am not sure if there is any UAV parameters that need to be set to allow MAV_CMD_DO_SET_MODE to function more like Set_Mode or if there is some issue with the library or something I am just doing wrong in general.

Has anyone else had any trouble with setting the CUSTOM_MODE using MAV_CMD_DO_SET_MODE?  Can you get its value to set to anything other than 0 (in Heartbeat)?  Thanks for the help.

benbarkay, 2019-01-24:
Which autopilot firmware and version are you working against?

Enlade, 2019-01-24:
Yea, it is probably not the latest stable version.  Let me make sure it is updated to the last Ardupilot version and get back to you.  Thanks.

dairoca90, 2020-11-15:
how did you send de SET_MODE im having issues with that

KKJava1, 2025-06-06:
> 你是怎么发送 SET_MODE 的？我遇到了问题
哥，您现在是怎么发送MAV_CMD_DO_SET_MODE。我的对接文档是需要传递
`
COMMAND_LONG {target_system : 1, target_component : 1, command : 176, confirmation : 0, param1 :1 , param2 : 0x30000, param3 : 0.0, param4 : 0, param5 : 0.0, param6 : 0.0, param7 : 0.0}
`
但是会转换成10进制，一直转模式被拒绝