dronefleet/mavlink #22: Message #269 VideoStreamInformation never receive

SilvainIADYS, 2019-01-30:
Hi,
I receive all my message except the #269 VideoStreamInformation message.
The low level (MavlinkPacket) receive it but not with MavlinkMessage.

Do you know why ?
Thanks for your reply

benbarkay, 2019-01-30:
If you could provide some bytes then that would be very helpful :)

SilvainIADYS, 2019-01-31:
Yes, of course. I using now 1.1.4 version.

When I use MavlinkPacketReader :
`I/System.out: MavlinkPacket{versionMarker=253, incompatibleFlags=0, compatibleFlags=0, sequence=27, systemId=1, componentId=100, messageId=269, payload=[0, 0, 32, 65, 0, -32, -91, 1, 1, 0, -48, 2, -32, 1, 0, 0, 0, 0, 1, 1, 2, 99, 97, 109, 101, 114, 97, 32, 102, 114, 111, 110, 116, 97, 108, 101], checksum=27381, signature=[]}`

When I break down the message, I see that there is not enough byte to encode the name and the uri (even with data truncation). But I do not know if it's important.

Here is the C code to send the message :
`mavlink_message_t msgRep;
  char name[32] = "camera frontale";
  char uri[160] = "";
  mavlink_msg_video_stream_information_pack(paramIdSysMavlink, paramIdCompMavlink, &msgRep, 1, 1, VIDEO_STREAM_TYPE_TCP_MPEG, VIDEO_STREAM_STATUS_FLAGS_RUNNING, FRAMERATE_MAX, 720, 480, (720 * 480 * FRAMERATE_MAX * 8), 0, 0, name, uri);`

Thanks for your help

SilvainIADYS, 2019-01-31:
Hi,
The message was shorter because Mavlink cut the zeros at the end and as uri (last field) was empty, it was not reconciled.
Any way, even with full data, the message is not received by MavlinkMessage.

Here is the log :
`I/System.out: MavlinkPacket{versionMarker=253, incompatibleFlags=0, compatibleFlags=0, sequence=23, systemId=1, componentId=100, messageId=269, payload=[0, 0, 32, 65, 0, -32, -91, 1, 1, 0, -48, 2, -32, 1, 0, 0, 0, 0, 1, 1, 2, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97, 97], checksum=27841, signature=[]}`

benbarkay, 2019-01-31:
I'll take a closer look at this over the weekend, however if what you say is true and there isn't enough space in the payload to contain the full message, then it's likely that what ends up happening is that the packet is dropped for invalid checksum -- Sender generates checksum based on full data, and receiver generates a mismatching one based on partial data.

Do you receive the message when encoding a shorter url/name that does fit a packet?

SilvainIADYS, 2019-01-31:
There is enough place in the payload and I get the message in full.
What I wanted to say is that mavlink removes the zeros at the end of the payload (see https://mavlink.io/en/guide/serialization.html#payload_truncation) and therefore the uri when it is empty (filled with zero).

In both cases, the packet is unrecognized.

benbarkay, 2019-01-31:
I understand. It would help significantly if you could record some bytes which contain the message into a file and upload it here, this way I'd have a quick way of reproducing the exact issue once I'm on it

benbarkay, 2019-02-01:
@SilvainIADYS You can also mark your own fields as extensions, in which case these would be added at the end of the packet and not create CRC discrepancies as they wouldn't change the message's CRC extra

benbarkay, 2019-02-01:
Looks like `VideoStreamInformation` has been modified 21 days ago at commit:
https://github.com/mavlink/mavlink/commit/84d1bcb94f76c490c3c66bddb6fe174ef65594c2#diff-ceacc1e2390186650146a45be8fcb691

Will make a release with updated dialects shortly :)
My bad with the previous two responses.

SilvainIADYS, 2019-02-04:
Tanks for your help. I will try with the 1.1.5 release.
I keep in mind that this message is not definitive.

benbarkay, 2019-02-06:
Has this been resolved?

SilvainIADYS, 2019-02-06:
Yes, it's perfect.
Thanks for your help

closed 2019-02-06 by SilvainIADYS