dronefleet/mavlink #23: tips for begginers

mateustormin, 2019-01-30:
First of all, thanks for all the effort and dedication put on this wonderful library.
I am a begginer on programming for mavlink, though I have already made some changes on a program that ran on a RedBear as companion computer for a 3DR iris+ a long time ago, but then the code was already written.

I am trying to develop a simple plan uploader for a ArduPilot drone that I don´t have yet, but I was anxious and wanted to get started already. So I am running a SITL drone and connecting it using the library.

I can get the heartbeat messages fine with this setup and a lot of other messages Im not interested in at the  moment, but I was trying to follow the protocol (https://mavlink.io/en/services/mission.html#uploading_mission) for mavlink upload a flight plan but I stoped at the first part because when I send a MissionCount message I get no response ever from the SITL.

So I decided to send the count and ignore the ack and trying to send a MissionItem message with one simple coordinate just to test it and I get a MAV_MISSION_ERROR back, how should I know whats the error?

this is the code for my MissionCount message
`MissionCount count = MissionCount.builder()
							.count(1)
							.missionType(MavMissionType.MAV_MISSION_TYPE_MISSION)
							.targetSystem(systemId)
							.targetComponent(componentId)
							.build();`
and this is the code for my MissionItem message:
`MissionItem item = MissionItem.builder()
							.seq(0)
							.frame(MavFrame.MAV_FRAME_GLOBAL)
							.command(MavCmd.MAV_CMD_NAV_WAYPOINT)
							.missionType(MavMissionType.MAV_MISSION_TYPE_MISSION)
							.current(1)
							.autocontinue(1)
							.param1(0f)//Hold time in decimal seconds
							.param2(0.2f)//Acceptance radius in meters
							.param3(0f)//if > 0 radius in meters to pass by WP
							.param4(0f)
							.x(48.995098f)
							.y(9.290475f)
							.z(215f)
							.build();`

can you point me in the right direction as to what im doing wrong?

Best regards

benbarkay, 2019-01-30:
I can think of a couple of things off the top of my head:
1. Are you sure that the system ID and component ID are correct? These should be the MAVs, not your own. You get these from the MAVs messages -- `MavlinkMessage#getOriginSystemId()` and `MavlinkMessage#getOriginComponentId()`.
2. You must only send mission items as response to `MissionRequest` and `MissionRequestInt` messages that the autopilot sends you, with either `MissionItem` or `MissionItemInt` respectively. Some autopilots support one but not the other.

mateustormin, 2019-01-30:
Thank you very much, your TopOfMind idea was the right one. I can now upload missions, I thought I saw somewhere that the system id and component id were locked at 255,0 for some reason.

Best Regards

benbarkay, 2019-01-30:
You're welcome

closed 2019-01-30 by benbarkay