dronefleet/mavlink #24: Message serializer always returning empty byte[]

zacpullen, 2019-03-08:
Thanks for the great library!

Having a few issues with serialization of the message before adding it to the mavlink packet to be sent.

Messages are being sent and received successfully by the client, however they have an empty body.

`HEARTBEAT {type : 0, autopilot : 0, base_mode : 0, custom_mode : 0, system_status : 0, mavlink_version : 0} 
`
Using the code as given in the example
```
int systemId = 255;
int componentId = 0;
Heartbeat heartbeat = Heartbeat.builder()
          .type(MavType.MAV_TYPE_GCS)
          .autopilot(MavAutopilot.MAV_AUTOPILOT_INVALID)
          .systemStatus(MavState.MAV_STATE_UNINIT)
          .mavlinkVersion(3)
          .build()

// Write an unsigned heartbeat
connection.send2(systemId, componentId, heartbeat);
```

Inside of connection.send2() the heartbeat is serialized  before it is added to the mavlink packet using:
```
byte[] serializedPayload = serializer.serialize(heartbeat);
```

This then implements ReflectionPayloadSerializer to return an array of bytes which is always returning empty. Looking into the problem I looked during debug and found that  the payload length was always returning as 0  meaning that the created array was of 0 size (byte[0])

See screenshot

[image]

```

public class ReflectionPayloadSerializer implements MavlinkPayloadSerializer {

    private static final WireFieldInfoComparator wireComparator = new WireFieldInfoComparator();

    @Override
    public byte[] serialize(Object message) {
        Class<?> messageClass = message.getClass();
        if (!messageClass.isAnnotationPresent(MavlinkMessageInfo.class)) {
            throw new IllegalArgumentException(messageClass.getName() + " is not annotated with @MavlinkMessageInfo");
        }

        int payloadLength = Arrays.stream(messageClass.getMethods())
                .filter(m -> m.isAnnotationPresent(MavlinkFieldInfo.class))
                .map(m -> m.getAnnotation(MavlinkFieldInfo.class))
                .mapToInt(f -> f.unitSize() * Math.max(1, f.arraySize()))
                .sum();
        if (payloadLength > 253) {
            throw new IllegalStateException("payload length > 253 for message" + messageClass.getName());
        }
        byte[] payload = new byte[payloadLength];  ---->> (payloadLength returning as 0 and hence  byte[] payload = new byte[0]; )

```
Any insight into the fix of this would be greatly appreciated!

Thanks in advanced,
Zac

benbarkay, 2019-03-08:
Running a quick test with `1.1.5` using a `1.8` JVM I was not able to reproduce this.
What versions of the library and JRE are you running?

zacpullen, 2019-03-08:
Hi Ben, 

The issue was occurring on an android app for a project that I was adding functionality to. I have repeated the process on a fresh app and also on a stand alone application and I was not able to replicate the result either. 

It seems that there is something fishy happening within the existing project.  Both the apps were running 1.1.5 and and 1.8.

Closing the thread as it seems that this is not a mavlink issue.

Thanks for the response and sorry for the false alarm.

Zac

closed 2019-03-08 by zacpullen