dronefleet/mavlink #25: Setting Rally Points on Copter 3.6

mateustormin, 2019-03-08:
I implemented a download and upload of missions with this great library and all went well, until I find out that currently released version of ArduCopter 3.6 does not support uploads of Rally Points using the mission protocol.

On the arducopter forum I found out that I have to set a parameter to the rally point size and then "push the rally points using RALLY_POINT" but I did not find a way to do this in dronefleet.io. Can you give me a hint on how can I upload rally points to ArduCopter 3.6 using your library please?

Thanks in advance.

benbarkay, 2019-03-09:
I have not done this myself, but it appears that this is what you would have to do based on what you described:
1. Send a `ParamSet` message which would set some parameter to the total amount of rally points that you would like to upload
2. Send `RallyPoint` messages corresponding to your rally points.

It doesn't seem like dronefleet/mavlink would be limiting you in any way

mateustormin, 2019-03-09:
I'm sorry if I caused the wrong impression, I was not stating that the
library would be limiting me, I was just trying to figure out how to send
the RallyPoint messages.
It turns out that I was looking for the message in the wrong package. The
RallyPoint is in the ardupilotmega package and I was only looking on the
common. My mistake, but it could help others that has the same problem that
I did.

Thanks for your help

Best Regards,

Mateus Tormin
GTP Automation
mateus@gtpautomation.com <mateus@gtptecnologia.com>
Mobile:    +55 11 94349 7163
Landline: +55 11 3721 6560

benbarkay, 2019-03-09:
No worries, you're welcome :)

closed 2019-03-09 by benbarkay