dronefleet/mavlink #26: Mavlink2Message methods cannot be found

MattAuto28, 2019-03-18:
I'm currently importing quite a bit of the dronefleet mavlink library into my java project. It has been working well for Mavlink 1 messages, but when attempting to implement Mavlink2Message, I cannot use any of the methods from Mavlink2Message.

The fact that I can successfully cast to Mavlink2Message, but cannot use any Mavlink2Message method, makes me believe that it is not on my end. Not only can I not use isSigned(), but I can't use any Mavlink2Message method.

It fails to compile with this error:
```
 error: cannot find symbol
                        if (message2.isSigned()) {
                                    ^
  symbol:   method isSigned()
  location: variable message2 of type Mavlink2Message
```

I follow the same method shown in the primary readme:

```
MavlinkMessage message;
    while ((message = connection.next()) != null) {
        if (message instanceof Mavlink2Message) {
            Mavlink2Message message2 = (Mavlink2Message)message;
            
            if (message2.isSigned()) {
                         //dosomething
                } else {
                    // etc
                }
            } else {
                // etc
            }
        } else {
           // etc
        }
    }
```

And, from dronefleet, I import the following:

```
import io.dronefleet.mavlink.MavlinkConnection;
import io.dronefleet.mavlink.MavlinkMessage;
import io.dronefleet.mavlink.Mavlink2Message;
import io.dronefleet.mavlink.common.Altitude;
import io.dronefleet.mavlink.common.Attitude;
import io.dronefleet.mavlink.common.AttitudeQuaternion;
import io.dronefleet.mavlink.common.AttitudeTarget;
import io.dronefleet.mavlink.common.BatteryStatus;
import io.dronefleet.mavlink.common.CommandAck;
import io.dronefleet.mavlink.common.CommandLong;
import io.dronefleet.mavlink.common.EstimatorStatus;
import io.dronefleet.mavlink.common.ExtendedSysState;
import io.dronefleet.mavlink.common.GlobalPositionInt;
import io.dronefleet.mavlink.common.GpsRawInt;
import io.dronefleet.mavlink.common.Heartbeat;
import io.dronefleet.mavlink.common.HighresImu;
import io.dronefleet.mavlink.common.HomePosition;
import io.dronefleet.mavlink.common.LocalPositionNed;
import io.dronefleet.mavlink.common.ManualControl;
import io.dronefleet.mavlink.common.MavAutopilot;
import io.dronefleet.mavlink.common.MavCmd;
import io.dronefleet.mavlink.common.MavMissionType;
import io.dronefleet.mavlink.common.MavMode;
import io.dronefleet.mavlink.common.MavState;
import io.dronefleet.mavlink.common.MavType;
import io.dronefleet.mavlink.common.MissionAck;
import io.dronefleet.mavlink.common.MissionClearAll;
import io.dronefleet.mavlink.common.MissionCount;
import io.dronefleet.mavlink.common.MissionSetCurrent;
import io.dronefleet.mavlink.common.MissionCurrent;
import io.dronefleet.mavlink.common.MissionItem;
import io.dronefleet.mavlink.common.MissionRequest;
import io.dronefleet.mavlink.common.MissionRequestInt;
import io.dronefleet.mavlink.common.ParamRequestRead;
import io.dronefleet.mavlink.common.ParamSet;
import io.dronefleet.mavlink.common.ParamValue;
import io.dronefleet.mavlink.common.Ping;
import io.dronefleet.mavlink.common.PositionTargetGlobalInt;
import io.dronefleet.mavlink.common.PositionTargetLocalNed;
import io.dronefleet.mavlink.common.ServoOutputRaw;
import io.dronefleet.mavlink.common.SysStatus;
import io.dronefleet.mavlink.common.VfrHud;
import io.dronefleet.mavlink.common.Vibration;
```

benbarkay, 2019-03-18:
Which version of dronefleet/mavlink are you using? Also, what do you mean by importing? Are you using the maven central coordinates with a build system like gradle or maven or are you going through some other process?

MattAuto28, 2019-03-19:
Importing directly as packages from io.dronefleet.mavlink such as, 
```
import io.dronefleet.mavlink.MavlinkConnection;
import io.dronefleet.mavlink.MavlinkMessage;
import io.dronefleet.mavlink.Mavlink2Message;
```
Using the latest version of dronefleet + Mavlink.

benbarkay, 2019-03-19:
This reveals what you have been doing in your class however does not explain how dronefleet/mavlink has gotten to your classpath or from where, and whether or not you are building it yourself or are retrieving already built binaries from maven central.

It appears that you're not using a known IDE like Intellij IDEA or Eclipse seeing as you are doing manual import labor. I highly recommend you give one of these IDEs a chance, personally vouching for Intellij IDEA. I'd suggest you try creating a gradle or maven project and use the maven central coordinates to include dronefleet/mavlink as a dependency. Otherwise, it will be tough to extract the relevant information to reproduce the situation you're in.

rohawa, 2019-03-22:
I'm working on the same project as MattAuto28, We're building out MavLink using gradle and implementing it into our application. We are not pulling it from Maven. For our use case, using IDEA or IntelliJ isn't compatible.  

I think our bigger issue might be related to the version of Mavlink being used (1.0.13), which seems to have had issues with Mavlink2 messages (fix:14). Does updating to 1.0.14 fix those issues or should we be using the latest build of Mavlink? I have tested 1.1.5 with our application but it breaks our current working code that is compatible with 1.0.13. It seems that there were big changes in the shift from 1.0.1x to 1.1.x

benbarkay, 2019-03-22:
@rohawa `1.0.14` or `1.0.15` might work out for you, however I'd advise to upgrade to `1.1.5` instead as it includes a lot of defect fixes and has been more thoroughly tested. It's a relatively small API change. If you're using protocol version 1 then upgrading to `1.1.5` shouldn't be a big refactor for you, you would simply have to call `send1` instead of `send`.

rohawa, 2019-03-25:
I thought there were changes made to the Mavlink2Message constructor between 1.0 and 1.1. That seems to be where our build throws errors. It looks like you have to build the Message using the packet and payload vs. IDs, Flags, and payload. 

I wouldn't have noticed the change to send2, thanks for pointing that out!

benbarkay, 2019-03-26:
@rohawa in `1.1.x` you do not have to construct `MavlinkMessage` or `Mavlink2Message` instances yourself at all. Instead, calls to `send1` or `send2` accept the payload and other parameters, and constructs and sends the message instances for you.

MattAuto28, 2019-03-28:
Great - will have to change some of the way my code interacts with the new methodology in the newer version of Dronefleet, but it looks like it should work now.

closed 2019-03-28 by MattAuto28