dronefleet/mavlink #28: Support default dialects instead of relying heartbeat to detect dialects

jxltom, 2019-03-27:
Some systems may not send heartbeat during communication at all. It will be nice to add default dialects as a list to parse packages, instead of using ```systemDialects```.

benbarkay, 2019-03-27:
This is a problem that should definitely be worked on.
I think that a system's dialect could be progressively inferred by sampling the input and matching it against `dialects`. Perhaps it's also a good idea to allow users to affix specific system IDs to dialects.

What do you think?

jxltom, 2019-03-28:
I made a PR https://github.com/dronefleet/mavlink/pull/30 to solve this.

benbarkay, 2019-03-29:
@jxltom I am still doubtful that this would be the right way to go about this. 

For instance, imagining two autopilots of different dialects, which have some colliding message IDs of different versions, sharing the same connection with the GCS -- In that case there would be no chance to resolve the correct dialect with default dialects.

Also, figuring the dialect of a system not only helps with understanding incoming messages, but to also understand what that system supports and what can be sent to it. So users of `MavlinkConnection` should have a way to retrieve that dialect, rather than it being transient state.

Please let me know if you have contentions in regards to the above. I appreciate you taking the time to open the issue and make a PR.

closed 2020-04-02 by benbarkay

jxltom, 2020-04-03:
This consideration is more comprehensive and thanks for the patch.

I will keep updating if I encounter any issues. Thanks.