dronefleet/mavlink #29: Any command execution fails

marioarrigonineri, 2019-03-27:
Hi,
I'm trying to implement an autonomous driver using java dronevlett.mavlink client library.
I cannot understand the correct set of commands I have to send to PX4 to takeoff.
I'm working on a local windows machine with PX4 and JMAV simulator.
Communication works fine, but I always receive a MAV_RESULT_FAILED.

Here is my simplest example:

		UdpConnector udp = new UdpConnector(InetAddress.getLocalHost(), 14570, 14550);
		final MavlinkConnection connection = MavlinkConnection.create(udp.getInputStream(), udp.getOutputStream());
		new Thread() {
			@Override
			public void run() {
				while (true) {
					try {
						connection.send1(255, 0,
								Heartbeat.builder().type(MavType.MAV_TYPE_GCS).autopilot(MavAutopilot.MAV_AUTOPILOT_PX4)
										.systemStatus(MavState.MAV_STATE_UNINIT).mavlinkVersion(1).build());
						Thread.sleep(1000);
					} catch (Exception e) {
						e.printStackTrace();
					}
				}
			}
		}.start();
		connection.send1(255, 0, CommandInt.builder().command(MavCmd.MAV_CMD_COMPONENT_ARM_DISARM).param1(1).build());

		MavlinkMessage message;
		while ((message = connection.next()) != null) {
			Object p = message.getPayload();
			if (p instanceof Heartbeat || p instanceof CommandAck)
				System.out.println("" + message.getSequence() + " --> " + p);
		}

the output is:

67 --> CommandAck{command=EnumValue{value=400, entry=MAV_CMD_COMPONENT_ARM_DISARM}, result=EnumValue{value=4, entry=MAV_RESULT_FAILED}, progress=0, resultParam2=0, targetSystem=0, targetComponent=0}
176 --> Heartbeat{type=EnumValue{value=2, entry=MAV_TYPE_QUADROTOR}, autopilot=EnumValue{value=12, entry=MAV_AUTOPILOT_PX4}, baseMode=EnumValue{value=81, entry=null}, customMode=65536, systemStatus=EnumValue{value=3, entry=MAV_STATE_STANDBY}, mavlinkVersion=3}

cannot understand where is the problem.. is there any specific handshare or command sequence?

thanks

Mario

benbarkay, 2019-03-27:
@marioarrigonineri have you tried using `CommandLong` instead?

marioarrigonineri, 2019-03-28:
thank you..
As a first test I'm working with px4 toolchain on windows.. I guess someone already used it as a test environment.
I'll check further documentation, but I see few details. Cannot find any "sequence of commands from zero to takeff".
thanks for any support

marioarrigonineri, 2019-03-29:
I tried using CommandLong.. same result :(

benbarkay, 2019-03-29:
Are you printing out `Statustext` messages? These sometimes print out human readable messages when something's wrong, it might reveal the problem if the command is correct but the vehicle's state prevents from arming.

Also, I know in my tests with ardupilot firmware that I must first set mode to guided. This could be the problem if compass is calibrated and there's GPS 3D fix.

benbarkay, 2019-03-29:
Also, the modes are set using the `customMode` portion of either `SetMode` or `MAV_CMD_DO_SET_MODE` depending on your firmware. So it's not as straightforward as setting the mode to something in `MavMode`, at least that is the case in ardupilot.

benbarkay, 2019-05-01:
Closed for inactivity

closed 2019-05-01 by benbarkay