dronefleet/mavlink #33: MavlinkSerializationException for supported field type: BigInteger

jameslabundis, 2019-08-12:
I'm currently using the Ping protocol to send ping a message to a connected drone. There seems to be a problem with serializing the BigInteger timeUsec in the Ping.java. The stack trace says MavlinkSerializationException: unrecognized field type java.math.BigInteger, yet the source of that exception, ReflectionPayloadSerializer.java, does handle BigInteger types. See stack trace below:

> io.dronefleet.mavlink.serialization.MavlinkSerializationException: unrecognized field type java.math.BigInteger
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadSerializer.lambda$serialize$6(ReflectionPayloadSerializer.java:73)
	at io.dronefleet.mavlink.serialization.payload.reflection.-$$Lambda$ReflectionPayloadSerializer$MYJl8X3lgsDBIeb9xszPJjyTJJk.accept(Unknown Source:10)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(ForEachOps.java:184)
	at java.util.stream.-$$Lambda$Abl7XfE0Z4AgkViLas9vhsO9mjw.accept(Unknown Source:2)
	at java.util.ArrayList.forEach(ArrayList.java:1262)
	at java.util.stream.SortedOps$RefSortingSink.end(SortedOps.java:390)
	at java.util.stream.Sink$ChainedReference.end(Sink.java:259)
	at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:483)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:472)
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:151)
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:133)
	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:235)
	at java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:419)
	at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadSerializer.serialize(ReflectionPayloadSerializer.java:49)
	at io.dronefleet.mavlink.MavlinkConnection.send(MavlinkConnection.java:308)

I build a Mavlink message containing the ping in the code below, but the exception does not occur until I actually send this message over the MavlinkConnection:
`new MavlinkMessage<>(255, 0, Ping.builder()
            .timeUsec(BigInteger.valueOf(System.currentTimeMillis()))
            .seq(Long.valueOf(seq))
            .targetSystem(droneId)
            .targetComponent(0)
            .build())`

I looked at where the MavlinkSerializationException is thrown in ReflectionPayloadSerializer.java:78 and there is a check for the BigInteger type before throwing that exception.

benbarkay, 2019-08-14:
@jameslabundis which version are you using?

jameslabundis, 2019-08-19:
I was using an old version from last November, which did not support this field type. Was able to fix this just by upgrading the build version. My bad. Closing this issue.

closed 2019-08-19 by jameslabundis