dronefleet/mavlink #35: Unable to get any kind of data from UDP connection

gadenis, 2019-10-18:
Hi,
I have an issue trying to get data with UDP connection to an arducopter >= 3.7.

I'm following these steps:
- create mavlink connection
- start to send CGS Heartbeat
- request Data Stream
- get messages from mavlinkConnection

I set up some wrapper and functions but these steps can be simplified in this way.

---create mavlink connection
```
Socket socket = new Socket(host, port, false);
socket.setSoTimeout(10000);

MavlinkConnection connection = MavlinkConnection.builder(socket.getInputStream(), socket.getOutputStream())
          .dialect(MavAutopilot.MAV_AUTOPILOT_GENERIC, new StandardDialect())
          .dialect(MavAutopilot.MAV_AUTOPILOT_ARDUPILOTMEGA, newArdupilotmegaDialect()).build();

```

- start to send CGS Heartbeat (sent every second)
```
      Heartbeat msg = Heartbeat.builder()
          .type(MavType.MAV_TYPE_GCS)
          .autopilot(MavAutopilot.MAV_AUTOPILOT_INVALID)
          .systemStatus(MavState.MAV_STATE_UNINIT)
          .mavlinkVersion(3)
          .build();
      try {
        connection.send2(255, 1, msg);
      } catch (IOException e) {
        Logger.logException(TAG, e);
      }
```

- request Data Stream
```
int heartbeatStreamId = 0;
int batteryStatusStreamId = 147;
CommandLong hbCmd = commandLongBuilder.command(MavCmd.MAV_CMD_SET_MESSAGE_INTERVAL).param1(heartbeatStreamId).param2(0).targetSystem(0).targetComponent(0).build();
CommandLong batteryCmd = commandLongBuilder.command(MavCmd.MAV_CMD_SET_MESSAGE_INTERVAL).param1(heartbeatStreamId).param2(0).targetSystem(0).targetComponent(0).build();
connection.send2(255, 1, hbCmd);
connection.send2(255, 1, batteryCmd);
```

- get messages from mavlinkConnection
```
      while (mavlinkConnection != null) {
        MavlinkMessage message = null;
        try {
          message = mavlinkConnection.next();
          ......
        } catch (Throwable e) {
          message = null;
          ......
        }
      }
```

I never get any kind of message from this connection, I always get timeout.
What am I doing wrong?

Is there any issue in the create of mavlink connection?
Do I have to do something else to start the mavlink connection in addition to sending the heartbeat and requesting the data stream?

Regarding the connection, I know that this kind of socket constructor is deprecated but I don't know any other way to create a mavlink connection with a UDP socket.
Can you suggest a better one?

Many thanks in advance

benbarkay, 2019-10-18:
Hi there.

`Socket` uses TCP, so it's likely failing to communicate with your UDP destination.
 To make UDP connections you will have to look into `DatagramSocket`.
Since DatagramSocket doesn't provide input/output streams, you will also need to make these for `MavlinkConnection` to consume. I suggest using `PipedInputStream` and `PipedOutputStream`.

Good luck!

gadenis, 2019-10-19:
Hi Ben,
many thanks for your reply.

I saw your suggestion in the **Overview** but honestly, I didn't find any way to get PipedInputStream / PipedOutputStream from a DatagramSocket.
I also looked for examples but without any success.

Can you please suggest how to get it?

Thanks

benbarkay, 2020-03-30:
This is a Java issue but I suppose a ready-made solution would be nice to have because of the friction it creates in terms of starting to use this with existing UDP-based tools like mavproxy. Perhaps this can be implemented in a separate library.

Here's a snippet that might help for now. I only tested this lightly, but it may help with direction:
```java
// The address we listen on to receive packets
SocketAddress localAddress = new InetSocketAddress(
        "127.0.0.1", 1234);

// The address to which we send packets
SocketAddress remoteAddress = new InetSocketAddress(
        "127.0.0.1", 4321);

// This is the size of our buffer. 65535 should be able to carry any
// UDP packet of a standard length.
int bufferSize = 65535;

// This binds to the listen address, we can later use this datagram socket
// to receive packets and mark packets with source addresses.
DatagramSocket datagramSocket = new DatagramSocket(localAddress);

// The input/output streams that will be exposed for use
PipedInputStream udpIn = new PipedInputStream();
OutputStream udpOut = new OutputStream() {
    // What we're doing here is we're basically buffering up writes to the output stream,
    // and then sending the data as a UDP packet when flushing.
    
    final byte[] buffer = new byte[bufferSize];
    int position = 0;

    @Override
    public void write(int i) throws IOException {
        write(new byte[] { (byte)i }, 0, 1);
    }

    @Override
    public synchronized void write(byte[] b, int off, int len) throws IOException {
        // If the buffer is full, we flush
        if ((position + len) > buffer.length) {
            flush();
        }
        System.arraycopy(b, off, buffer, position, len);
        position += len;
    }

    @Override
    public synchronized void flush() throws IOException {
        DatagramPacket packet = new DatagramPacket(buffer, 0, position, remoteAddress);
        datagramSocket.send(packet);
        position = 0;
    }
};

// We connect `udpIn` to a corresponding PipedOutputStream (`appOut`). Data that we write to
// `appOut` will be available to read from `udpIn`.
// We do this here rather than within the reading thread so that `udpIn` can be used immediately.
PipedOutputStream appOut = new PipedOutputStream(udpIn);

// We create a thread pool with 1 thread which we use for receiving packets.
// We keep reading incoming packets and pushing them into `appOut`, so that they
// are available to read from `udpIn`.
ExecutorService service = Executors.newSingleThreadExecutor();
service.execute(() -> {
    try {
        DatagramPacket packet = new DatagramPacket(new byte[bufferSize], bufferSize);
        while (!datagramSocket.isClosed()) {
            datagramSocket.receive(packet);
            appOut.write(packet.getData(), packet.getOffset(), packet.getLength());
            appOut.flush();
        }
    } catch (IOException e) {
        e.printStackTrace();
    } finally {
        try {
            appOut.close();
        } catch (IOException e) {
            e.printStackTrace();
        }
        if (!datagramSocket.isClosed()) {
            datagramSocket.close();
        }
        if (!service.isShutdown()) {
            service.shutdownNow();
        }
    }
});

// We are now left with udpIn and udpOut, with which we can use to read/write
```

Here's a simple relay example (writes whatever it receives to the remote endpoint):
```java
byte[] receiveBuffer = new byte[bufferSize];
int readLength;
while ((readLength = udpIn.read(receiveBuffer)) != -1) {
    udpOut.write(receiveBuffer, 0, readLength);
    udpOut.flush();
}
```

TurboTu, 2021-03-11:
@benbarkay
Hi, Ben
I'm working on a project based on mavlink java via udp.I tried your solution above, but still not wokring.I wonder is there any simple way to solve this?

OctavianIonel, 2021-04-15:
> This is a Java issue but I suppose a ready-made solution would be nice to have because of the friction it creates in terms of starting to use `dronefleet/mavlink` with existing UDP-based tools like mavproxy. Perhaps this can be implemented in a separate library.
> 
> Here's a snippet that might help for now. I only tested this lightly, but it may help with direction:
> 
> ```java
> // The address we listen on to receive packets
> SocketAddress localAddress = new InetSocketAddress(
>         "127.0.0.1", 1234);
> 
> // The address to which we send packets
> SocketAddress remoteAddress = new InetSocketAddress(
>         "127.0.0.1", 4321);
> 
> // The size we use for buffers. 65535 should be able to carry any
> // UDP packet of a standard length.
> int bufferSize = 65535;
> 
> // This binds to the listen address, we can later use this datagram socket
> // to receive packets and mark packets with source addresses when sending.
> DatagramSocket datagramSocket = new DatagramSocket(localAddress);
> 
> // The input/output streams that will be exposed for use. These are what you'd let
> // MavlinkConnection use.
> PipedInputStream udpIn = new PipedInputStream();
> OutputStream udpOut = new OutputStream() {
>     // We're buffering data that is written to the output stream,  and then sending the
>     // data as a UDP packet when flushing.
>     
>     final byte[] buffer = new byte[bufferSize];
>     int position = 0;
> 
>     @Override
>     public void write(int i) throws IOException {
>         write(new byte[] { (byte)i }, 0, 1);
>     }
> 
>     @Override
>     public synchronized void write(byte[] b, int off, int len) throws IOException {
>         // If the buffer is full, we flush
>         if ((position + len) > buffer.length) {
>             flush();
>         }
>         System.arraycopy(b, off, buffer, position, len);
>         position += len;
>     }
> 
>     @Override
>     public synchronized void flush() throws IOException {
>         DatagramPacket packet = new DatagramPacket(buffer, 0, position, remoteAddress);
>         datagramSocket.send(packet);
>         position = 0;
>     }
> };
> 
> // We connect `udpIn` to a corresponding PipedOutputStream (`appOut`). Data that we write to
> // `appOut` will be available to read from `udpIn`.
> // We instantiate/connect here rather than within the reading thread so that `udpIn` can be used immediately.
> PipedOutputStream appOut = new PipedOutputStream(udpIn);
> 
> // We create a thread pool with 1 thread which we use for receiving packets.
> // We keep reading incoming packets and pushing them into `appOut`, so that they
> // are available to read from `udpIn`.
> ExecutorService service = Executors.newSingleThreadExecutor();
> service.execute(() -> {
>     try {
>         DatagramPacket packet = new DatagramPacket(new byte[bufferSize], bufferSize);
>         while (!datagramSocket.isClosed()) {
>             datagramSocket.receive(packet);
>             appOut.write(packet.getData(), packet.getOffset(), packet.getLength());
>             appOut.flush();
>         }
>     } catch (IOException e) {
>         e.printStackTrace();
>     } finally {
>         try {
>             appOut.close();
>         } catch (IOException e) {
>             e.printStackTrace();
>         }
>         if (!datagramSocket.isClosed()) {
>             datagramSocket.close();
>         }
>         if (!service.isShutdown()) {
>             service.shutdown();
>         }
>     }
> });
> 
> // We are now left with udpIn and udpOut, with which we can use to read/write
> ```
> 
> Here's a simple relay example (writes whatever it receives to the remote endpoint):
> 
> ```java
> byte[] receiveBuffer = new byte[bufferSize];
> int readLength;
> while ((readLength = udpIn.read(receiveBuffer)) != -1) {
>     udpOut.write(receiveBuffer, 0, readLength);
>     udpOut.flush();
> }
> ```

Hello, 
I have some questions:
1) In this example with UDP, where are you using the Mavlink specific classes? Like MavlinkConnection?
Maybe an example with sending and another one with receiving will make a lot of sense. 
2) I am not sure to have understood exactly how it works Mavlink. It shouldn't be similar to how we serialize/deserialize JSON with Gson or Jettison or whatever?

benbarkay, 2021-04-15:
@OctavianIonel 

As for your first question, you would use `udpIn` and `udpOut` to construct a `MavlinkConnection` like you would with a `Socket`'s input/output streams:
```java
MavlinkConnection connection = MavlinkConnection.create(udpIn, udpOut);
```
You can then use `connection` as shown in the README examples.

To answer your second question, not exactly. 
Mavlink is not a generic object format like JSON is. It includes a generic packet format, but that's about where the similarity to JSON ends. To understand a message, you must resolve it against a specific dialect. I have tried to make this implementation as closely resembling to JSON as possible by allowing one to implement a dialect using reflection, but it's impossible to implement in the way that GSON can for JSON.

Additionally, there are some nuances related to the protocol which you would have to implement, which are related to the current state of the connection: sequencing, packet dropping, signing, dialect selection, etc.

If you want to implement these yourself, and deal with something more similar to json serialization/deserialization, then you can use `MavlinkPacketReader` to parse packets, and then use `ReflectionPayloadDeserializer` to deserialize it into an actual message object. You can take a look at the implementation of `MavlinkConnection` to see how that all works.

OctavianIonel, 2021-04-15:
> @OctavianIonel
> 
> As for your first question, you would use `udpIn` and `udpOut` to construct a `MavlinkConnection` like you would with a `Socket`'s input/output streams:
> 
> ```java
> MavlinkConnection connection = MavlinkConnection.create(udpIn, udpOut);
> ```
> 
> You can then use `connection` as shown in the README examples.
> 
> To answer your second question, not exactly.
> Mavlink is not a generic object format like JSON is. It includes a generic packet format, but that's about where the similarity to JSON ends. To understand a message, you must resolve it against a specific dialect. I have tried to make this implementation as closely resembling to JSON as possible by allowing one to implement a dialect using reflection, but it's impossible to implement in the way that GSON can for JSON.
> 
> Additionally, there are some nuances related to the protocol which you would have to implement, which are related to the current state of the connection, such as sequencing and dialect selection.
> 
> If you want to implement these yourself, and deal with something more similar to json serialization/deserialization, then you can use `MavlinkPacketReader` to parse packets, and then use `ReflectionPayloadDeserializer` to deserialize it into an actual message object. You can take a look at the implementation of `MavlinkConnection` to see how all of that works.

@benbarkay thanks for your answer.
As to close the loop and try it locally, do you know how can I configure a mavlink server in java? Thanks

benbarkay, 2021-04-15:
@OctavianIonel I've been using ardupilot's simulators successfully. If you want to implement your own server then you can simply use this library with a `ServerSocket` or a `DatagramSocket`. I won't get into detail of how to do these, as they are off-topic here, and there are many resources that do just that online.

OctavianIonel, 2021-04-16:
Hi @benbarkay 
I have configured a TCP server in python and with the example in the readme it works (I can get the heartbeat on the server).

Now I am dealing with the UDP server. My client code is the following, but nothing is received on the server side:

       int systemId = 255;
        int componentId = 0;

        //UDP
        try {
            // The address we listen on to receive packets
            SocketAddress localAddress = new InetSocketAddress(
                    "127.0.0.1", 1234);

            // The address to which we send packets
            SocketAddress remoteAddress = new InetSocketAddress(
                    "127.0.0.1", 14540);

            // The size we use for buffers. 65535 should be able to carry any
            // UDP packet of a standard length.
            int bufferSize = 65535;

            // This binds to the listen address, we can later use this datagram socket
            // to receive packets and mark packets with source addresses when sending.
            DatagramSocket datagramSocket = new DatagramSocket(localAddress);
            PipedInputStream udpIn = new PipedInputStream();

            OutputStream udpOut = new OutputStream() {
                // We're buffering data that is written to the output stream,  and then sending the
                // data as a UDP packet when flushing.

                final byte[] buffer = new byte[bufferSize];
                int position = 0;

                @Override
                public void write(int i) throws IOException {
                    write(new byte[] { (byte)i }, 0, 1);
                }

                @Override
                public synchronized void write(byte[] b, int off, int len) throws IOException {
                    // If the buffer is full, we flush
                    if ((position + len) > buffer.length) {
                        flush();
                    }
                    System.arraycopy(b, off, buffer, position, len);
                    position += len;
                }

                @Override
                public synchronized void flush() throws IOException {
                    DatagramPacket packet = new DatagramPacket(buffer, 0, position, remoteAddress);
                    datagramSocket.send(packet);
                    position = 0;
                }
            };

            // We connect `udpIn` to a corresponding PipedOutputStream (`appOut`). Data that we write to
// `appOut` will be available to read from `udpIn`.
// We instantiate/connect here rather than within the reading thread so that `udpIn` can be used immediately.
            PipedOutputStream appOut = new PipedOutputStream(udpIn);

// We create a thread pool with 1 thread which we use for receiving packets.
// We keep reading incoming packets and pushing them into `appOut`, so that they
// are available to read from `udpIn`.
            ExecutorService service = Executors.newSingleThreadExecutor();
            service.execute(() -> {
                try {
                    DatagramPacket packet = new DatagramPacket(new byte[bufferSize], bufferSize);
                    while (!datagramSocket.isClosed()) {
                        datagramSocket.receive(packet);
                        appOut.write(packet.getData(), packet.getOffset(), packet.getLength());
                        appOut.flush();
                    }
                } catch (IOException e) {
                    e.printStackTrace();
                } finally {
                    try {
                        appOut.close();
                    } catch (IOException e) {
                        e.printStackTrace();
                    }
                    if (!datagramSocket.isClosed()) {
                        datagramSocket.close();
                    }
                    if (!service.isShutdown()) {
                        service.shutdown();
                    }
                }
            });

            Heartbeat heartbeat = Heartbeat.builder()
                    .type(MavType.MAV_TYPE_GCS)
                    .autopilot(MavAutopilot.MAV_AUTOPILOT_INVALID)
                    .systemStatus(MavState.MAV_STATE_UNINIT)
                    .mavlinkVersion(3)
                    .build();
            MavlinkConnection connection = MavlinkConnection.create(udpIn, udpOut);
            connection.send2(systemId, componentId, heartbeat);
        } catch (IOException e) {
            e.printStackTrace();
        }

OctavianIonel, 2021-04-19:
@benbarkay I tried the code above with the UDP and it is not working, what could be the issue? Thanks for your help!

benbarkay, 2021-04-19:
In terms of reading there should not be an issue.
Try calling `udpOut.flush()` right after `connection.send`, so that packets
are sent out immediately, perhaps that is preventing you from seeing
traffic.

OctavianIonel, 2021-04-19:
Indeed this was the problem, now it works! Many thanks!

OctavianIonel, 2021-04-19:
So the situation is as follows:
1. I can send Mavlink message from the client to the server using UDP and TCP, the server will print a message in the console upon receiving
2. I can send Mavlink message from the server back to the client using TCP, but over UDP I don't receive anything in the client side. Could you help also here pls?

OctavianIonel, 2021-04-20:
@benbarkay to be more precise, it is not going any further after this line: datagramSocket.receive(packet);

OctavianIonel, 2021-04-20:
@benbarkay What could be the issue? Could you help me please?

OctavianIonel, 2021-09-16:
@benbarkay I have an issue with the frequency that the datagram packet is received (delayed comparing to the send).
Could it be because the datagram packet is not sent from a thread?
```
        OutputStream udpOut = new OutputStream() {
            // We're buffering data that is written to the output stream,  and then sending the
            // data as a UDP packet when flushing.

            final byte[] buffer = new byte[bufferSize];
            int position = 0;

            @Override
            public void write(int i) throws IOException {
                write(new byte[] { (byte)i }, 0, 1);
            }

            @Override
            public synchronized void write(byte[] b, int off, int len) throws IOException {
                // If the buffer is full, we flush
                if ((position + len) > buffer.length) {
                    flush();
                }
                System.arraycopy(b, off, buffer, position, len);
                position += len;
            }

            @Override
            public synchronized void flush() throws IOException {
                DatagramPacket packet = new DatagramPacket(buffer, 0, position, remoteAddress);
                datagramSocket.send(packet);
                position = 0;
            }
        };
```
while for the reading, we have a dedicated executor service:
```
        ExecutorService service = Executors.newSingleThreadExecutor();
        service.execute(() -> {
            try {
                DatagramPacket packet = new DatagramPacket(new byte[bufferSize], bufferSize);
                while (!datagramSocket.isClosed()) {
                    datagramSocket.receive(packet);
                    appOut.write(packet.getData(), packet.getOffset(), packet.getLength());
                    appOut.flush();
                }
            } catch (IOException e) {
                e.printStackTrace();
            } finally {
                try {
                    appOut.close();
                } catch (IOException e) {
                    e.printStackTrace();
                }
                if (!datagramSocket.isClosed()) {
                    datagramSocket.close();
                }
                if (!service.isShutdown()) {
                    service.shutdown();
                }
            }
        });
```

OctavianIonel, 2022-01-24:
Hi @benbarkay 
By using the library adapted for UDP, I get sometimes 

```
java.io.IOException: Read end dead
at java.io.PipedInputStream.checkStateForReceive(PipedInputStream.java:265)
at java.io.PipedInputStream.receive(PipedInputStream.java:229)
at java.io.PipedOutputStream.write(PipedOutputStream.java:149)
at TheConnection.lambda$setupTheConnection$0$ThekConnection(TheConnection.java:156)
```

my code is the following:
`       try {
            // This binds to the listen address, we can later use this datagram socket
            // to receive packets and mark packets with source addresses when sending.
            mDatagramSocket = new DatagramSocket(mClientAddress);
     
            mUdpIn = new PipedInputStream();
           
            mUdpOut = new OutputStream() {
                // We're buffering data that is written to the output stream,  and then sending the
                // data as a UDP packet when flushing.

                final byte[] buffer = new byte[bufferSize];
                int position = 0;

                @Override
                public void write(int i) throws IOException {
                    write(new byte[] { (byte)i }, 0, 1);
                }

                @Override
                public synchronized void write(byte[] b, int off, int len) throws IOException {
                    // If the buffer is full, we flush
                    if ((position + len) > buffer.length) {
                        flush();
                    }
                    System.arraycopy(b, off, buffer, position, len);
                    position += len;
                }

                @Override
                public synchronized void flush() throws IOException {
                    DatagramPacket packet = new DatagramPacket(buffer, 0, position, mHerelinkGroundAddress);
                    if (mDatagramSocket != null && !mDatagramSocket.isClosed()) {
                        mDatagramSocket.send(packet);
                    }
                    position = 0;
                }
            };

            // We connect `udpIn` to a corresponding PipedOutputStream (`appOut`). Data that we write to
            // `appOut` will be available to read from `udpIn`.
            // We instantiate/connect here rather than within the reading thread so that `udpIn` can be used immediately.
            mAppOut = new PipedOutputStream(mUdpIn);

            // We create a thread pool with 1 thread which we use for receiving packets.
            // We keep reading incoming packets and pushing them into `appOut`, so that they
            // are available to read from `udpIn`.
            mService = Executors.newSingleThreadExecutor();
            mService.execute(() -> {
                try {
                    DatagramPacket packet = new DatagramPacket(new byte[bufferSize], bufferSize);
                    while (mDatagramSocket != null && !mDatagramSocket.isClosed()) {
                        mDatagramSocket.receive(packet);
                        mAppOut.write(packet.getData(), packet.getOffset(), packet.getLength());
                        mAppOut.flush();
                    }
                } catch (IOException e) {
                    e.printStackTrace();
                }
            });

            mMavlinkConnection = MavlinkConnection
                    .builder(mUdpIn, mUdpOut)
                    .defaultDialect(new HerelinkMavDialect())
                    .build();

            //send a Heartbeat to the server
            Heartbeat heartbeat = Heartbeat.builder()
                    .type(MavType.MAV_TYPE_GCS)
                    .autopilot(MavAutopilot.MAV_AUTOPILOT_INVALID)
                    .systemStatus(MavState.MAV_STATE_UNINIT)
                    .mavlinkVersion(3)
                    .build();

            TimerTask timerTask = new TimerTask() {
                @Override
                public void run() {
                    try {
                        if (mMavlinkConnection != null) {
                            mMavlinkConnection.send2(Constants.SYSTEM_ID, Constants.COMPONENT_ID, heartbeat);
                        }
                        if (mUdpOut != null) {
                            mUdpOut.flush();
                        }
                    } catch (IOException e) {
                        if (mTimer != null) {
                            mTimer.cancel();
                            mTimer.purge();
                        }
                        e.printStackTrace();
                    }
                }
            };
            mTimer.schedule(timerTask, 0, 1000L);

        } catch (SocketException se) {
            if (mTimer != null) {
                mTimer.cancel();
                mTimer.purge();
            }
            se.printStackTrace();
        } catch (IOException e) {
            e.printStackTrace();
            if (mTimer != null) {
                mTimer.cancel();
                mTimer.purge();
            }
        }`

zeev-mindali, 2022-02-04:
@OctavianIonel did you mange to get it to work?
i also looking for udp solution with no success at this time...

zeev-mindali, 2022-02-04:
@benbarkay could you assist to a friend from israel, how to achive udp connection?

OctavianIonel, 2022-02-04:
> @OctavianIonel did you mange to get it to work?
> i also looking for udp solution with no success at this time...

Yes, I made it work with udp

zeev-mindali, 2022-02-04:
@OctavianIonel could you share the code please?
i trying to send commands to herelink rc which works on udp port 15442

benbarkay, 2022-02-04:
> @benbarkay could you assist to a friend from israel, how to achive udp connection?

What have you been trying so far? When it fails, what exceptions/errors/logs are you getting in the JVM/mavproxy/other software in the loop?

zeev-mindali, 2022-02-05:
in basic
i have apm 4.0.5, connected to herelink remote controller.
i want be able to change pwm servo value
in mavproxy, i am able to send : 
`message COMMAND_LONG 0 0 183 0 7 1500 0 0 0 0 0`

now i trying with udp connection to the rc, but don't getting any response and send command not effects

my code:
```
public class UDPmessage {
    private PipedInputStream udpIn;
    private Heartbeat heartbeat;
    private OutputStream udpOut;
    private DatagramSocket mDatagramSocket;

    public static void main(String[] args) {
        createConnection(255, 0);
    }

    private static void createConnection(int systemId, int componentId) {
        //UDP
        try {
            SocketAddress localAddress = new InetSocketAddress("192.168.50.238", 14552);
            SocketAddress remoteAddress = new InetSocketAddress("192.168.50.238", 14552);
            int bufferSize = 65535;
            DatagramSocket datagramSocket = new DatagramSocket(localAddress);
            PipedInputStream udpIn = new PipedInputStream();

            OutputStream udpOut = new OutputStream() {
                final byte[] buffer = new byte[bufferSize];
                int position = 0;
                @Override
                public void write(int i) throws IOException {
                    write(new byte[]{(byte) i}, 0, 1);
                }
                @Override
                public synchronized void write(byte[] b, int off, int len) throws IOException {
                    if ((position + len) > buffer.length) {
                        flush();
                    }
                    System.arraycopy(b, off, buffer, position, len);
                    position += len;
                }

                @Override
                public synchronized void flush() throws IOException {
                    DatagramPacket packet = new DatagramPacket(buffer, 0, position, remoteAddress);
                    datagramSocket.send(packet);
                    position = 0;
                }
            };

            PipedOutputStream appOut = new PipedOutputStream(udpIn);
            ExecutorService service = Executors.newSingleThreadExecutor();
            service.execute(() -> {
                try {
                    DatagramPacket packet = new DatagramPacket(new byte[bufferSize], bufferSize);
                    while (!datagramSocket.isClosed()) {
                        datagramSocket.receive(packet);
                        appOut.write(packet.getData(), packet.getOffset(), packet.getLength());
                        appOut.flush();
                    }
                } catch (IOException e) {
                    e.printStackTrace();
                } finally {
                    try {
                        appOut.close();
                    } catch (IOException e) {
                        e.printStackTrace();
                    }
                    if (!datagramSocket.isClosed()) {
                        datagramSocket.close();
                    }
                    if (!service.isShutdown()) {
                        service.shutdown();
                    }
                }
            });

            Heartbeat heartbeat = Heartbeat.builder()
                    .type(MavType.MAV_TYPE_GCS)
                    .autopilot(MavAutopilot.MAV_AUTOPILOT_INVALID)
                    .systemStatus(MavState.MAV_STATE_UNINIT)
                    .mavlinkVersion(3)
                    .build();
            MavlinkConnection connection = MavlinkConnection.create(udpIn, udpOut);
            udpOut.flush();
            connection.send2(systemId, componentId, heartbeat);
            //command
            CommandLong command = getCommand(5,1200);
            connection.send2(systemId, componentId, command);
        } catch (IOException e) {
            System.out.println("Error: " + e.getMessage());
        }
    }

    private static CommandLong getCommand(int servo, int pwm) {
        return CommandLong.builder()
                .command(MavCmd.MAV_CMD_DO_SET_SERVO)
                .confirmation(0)
                .param1(5).param2(1200).param3(0).param4(0).param5(0).param6(0).param7(0)
                .targetSystem(1).targetComponent(1)
                .build();
    }
}
```

zeev-mindali, 2022-02-05:
if you have any suggestion , it will be great

zeev-mindali, 2022-02-05:
i run mavsdk_server on port 50051 and connect it to my herelink rc.
getting a signal from the server:
```
[07:54:59|Info ] MAVSDK version: v1.0.8 (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:28)
[07:54:59|Info ] Waiting to discover system on udp://192.168.50.238:14552... (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\connection_initiator.h:20)
[07:54:59|Debug] Initializing connection to remote system... (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:482)
[07:55:00|Debug] New: System ID: 1 Comp ID: 1 (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:220)
[07:55:00|Debug] Component Autopilot (1) added. (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:360)
[07:55:00|Debug] New: System ID: 255 Comp ID: 190 (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:484)
[07:55:00|Debug] Component Unsupported component (190) added. (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:360)
[07:55:00|Debug] Discovered 1 component(s) (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:528)
[07:55:00|Info ] System discovered (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\connection_initiator.h:63)
[07:55:00|Info ] Server started (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\grpc_server.cpp:52)
[07:55:00|Info ] Server set to listen on 0.0.0.0:50051 (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\grpc_server.cpp:53)
[07:55:00|Debug] New: System ID: 42 Comp ID: 251 (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:484)
[07:55:00|Debug] Component Unsupported component (251) added. (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:360)
[07:55:00|Debug] Component Unsupported component (250) added. (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:360)
[07:55:00|Debug] Discovered 2 component(s) (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:528)
[07:55:00|Debug] Component Camera 1 (100) added. (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:360)
```

when sending heart beat, getting :
```
Socket[addr=/127.0.0.1,port=50051,localport=49941]
java.io.EOFException: End of stream
	at io.dronefleet.mavlink.protocol.MavlinkPacketReader.next(MavlinkPacketReader.java:47)
	at io.dronefleet.mavlink.MavlinkConnection.next(MavlinkConnection.java:238)
	at main.newTest.main(newTest.java:38)
```

zeev-mindali, 2022-02-05:
sent an explation in issue #70

zeev-mindali, 2022-02-06:
@OctavianIonel can you share the code for connection?

OctavianIonel, 2022-02-06:
@zeev-mindali I think you should send the heartbeat repetitively. Wrap that in a timer for example. In your code you are sending it only once.

zeev-mindali, 2022-02-20:
@OctavianIonel in tcp i getting and sending heart beat.
i using herelink RC, and i getting confused when i need UDP in and when UDP out.
should them both need to point to herelink server or one for RC and one for computer?

zeev-mindali, 2022-02-22:
@OctavianIonel it stuck on connection.next()
any suggestion?

moralkerim, 2022-02-25:
> This is a Java issue but I suppose a ready-made solution would be nice to have because of the friction it creates in terms of starting to use `dronefleet/mavlink` with existing UDP-based tools like [...]

Hi, I can receive messages with this example but can't seem to send a heartbeat to autopilot. I could be able to send heartbeat by tcp. What might be the problem? Would be great if you can help. Thanks.

swagatampanda3894, 2023-12-18:
after implement your code i got java.net.BindException: bind failed: EADDRNOTAVAIL (Cannot assign requested address) error ,

I have provide the android ip address , and port

gremsyQuanLe, 2023-12-18:
show your udp connection code so we can help ???

swagatampanda3894, 2023-12-18:
AppExecutors.getInstance().getDiskIO().execute(() -> {
            try {

                // The address we listen on to receive packets
                SocketAddress localAddress = new InetSocketAddress(
                        "192.168.144.12", UDP_PORT);

               // The address to which we send packets
                SocketAddress remoteAddress = new InetSocketAddress(
                        "192.168.144.12", UDP_PORT);

                // The size we use for buffers. 65535 should be able to carry any
                // UDP packet of a standard length.
                int bufferSize = 65535;

                // This binds to the listen address, we can later use this datagram socket
                // to receive packets and mark packets with source addresses when sending.
                DatagramSocket datagramSocket = new DatagramSocket(localAddress);

                   // The input/output streams that will be exposed for use. These are what you'd let
                   // MavlinkConnection use.
                PipedInputStream udpIn = new PipedInputStream();
                OutputStream udpOut = new OutputStream() {
                    // We're buffering data that is written to the output stream,  and then sending the
                    // data as a UDP packet when flushing.

                    final byte[] buffer = new byte[bufferSize];
                    int position = 0;

                    @Override
                    public void write(int i) throws IOException {
                        write(new byte[] { (byte)i }, 0, 1);
                    }

                    @Override
                    public synchronized void write(byte[] b, int off, int len) throws IOException {
                        // If the buffer is full, we flush
                        if ((position + len) > buffer.length) {
                            flush();
                        }
                        System.arraycopy(b, off, buffer, position, len);
                        position += len;
                    }

                    @Override
                    public synchronized void flush() throws IOException {
                        DatagramPacket packet = new DatagramPacket(buffer, 0, position, remoteAddress);
                        datagramSocket.send(packet);
                        position = 0;
                    }
                };

// We connect `udpIn` to a corresponding PipedOutputStream (`appOut`). Data that we write to
// `appOut` will be available to read from `udpIn`.
// We instantiate/connect here rather than within the reading thread so that `udpIn` can be used immediately.
                PipedOutputStream appOut = new PipedOutputStream(udpIn);

// We create a thread pool with 1 thread which we use for receiving packets.
// We keep reading incoming packets and pushing them into `appOut`, so that they
// are available to read from `udpIn`.
                ExecutorService service = Executors.newSingleThreadExecutor();
                service.execute(() -> {
                    try {
                        DatagramPacket packet = new DatagramPacket(new byte[bufferSize], bufferSize);
                        while (!datagramSocket.isClosed()) {
                            datagramSocket.receive(packet);
                            Toast.makeText(getApplicationContext(),"PACKETRECEIVED "+ packet.getLength(),Toast.LENGTH_SHORT).show();
                            appOut.write(packet.getData(), packet.getOffset(), packet.getLength());
                            appOut.flush();
                        }
                    } catch (IOException e) {
                        Log.e("ERRROR ","ERRORS1 "+ e);
                        e.printStackTrace();
                    } finally {
                        try {
                            appOut.close();
                        } catch (IOException e) {
                            e.printStackTrace();
                        }
                        if (!datagramSocket.isClosed()) {
                            datagramSocket.close();
                        }
                        if (!service.isShutdown()) {
                            service.shutdown();
                        }
                    }
                });

                // We are now left with udpIn and udpOut, with which we can use to read/write

                byte[] receiveBuffer = new byte[bufferSize];
                int readLength;
                while ((readLength = udpIn.read(receiveBuffer)) != -1) {
                    udpOut.write(receiveBuffer, 0, readLength);
                    udpOut.flush();
                }

            } catch (IOException e){
                Log.e("ERRROR ","ERRORS "+ e);

            }
this above code

swagatampanda3894, 2023-12-18:
The issue in port case , I am confused that in port that source and destination,

MrBoxed, 2024-05-23:
Here is the working UDP Connection sample code 

private void UDPConnection(){

    // ConnectionDataClass holding the IP & PORT of connection
        ConnectionData udpDataRef = getSocketData();
        if (udpDataRef == null){
                return;
        }

            // The address we listen on to receive/send packets
            SocketAddress remoteAddress = new InetSocketAddress(udpDataRef.getIpAddress(), udpDataRef.getPort());

            // The size we use for buffers. 65535 should be able to carry any
            // UDP packet of a standard length.
            int bufferSize = 65535;

            try {
                // This binds to the listen address, we can later use this datagram socket
                // to receive packets and mark packets with source addresses when sending.
                DatagramSocket udpSocket = new DatagramSocket();
                udpSocket.connect(remoteAddress);

                PipedInputStream udpIn = new PipedInputStream();

                OutputStream udpOut = new OutputStream() {

                    final byte[] buffer = new byte[bufferSize];
                    int position = 0;

                    @Override
                    public void write(int b) throws IOException {

                        write(new byte[]{(byte) b}, 0, 1);
                    }

                    @Override
                    public void write(byte[] b, int off, int len) throws IOException {

                        // if the buffer is full, we flush
                        if ((position + len) > (buffer.length)) {
                            flush();
                        }
                        System.arraycopy(b, off, buffer, position, len);
                        position += len;
                    }

                    @Override
                    public void flush() throws IOException {
                        DatagramPacket packet = new DatagramPacket(buffer, 0, position, remoteAddress);
                        udpSocket.send(packet);
                        position = 0;
                    }
                };

                // We connect `udpIn` to a corresponding PipedOutputStream (`appOut`). Data that we write to
                // `appOut` will be available to read from `udpIn`.
                // We instantiate/connect here rather than within the reading thread so that `udpIn` can be used immediately.
                PipedOutputStream appOut = new PipedOutputStream(udpIn);

                ExecutorService service = Executors.newSingleThreadExecutor();
                service.execute(() -> {
                    try {
                        DatagramPacket packet = new DatagramPacket(new byte[bufferSize], bufferSize);
                        while (!udpSocket.isClosed()) {
                            udpSocket.receive(packet);
                            appOut.write(packet.getData(), packet.getOffset(), packet.getLength());
                            appOut.flush();
                        }
                    } catch (IOException e) {
                        e.printStackTrace();
                    } finally {
                        try {
                            appOut.close();
                        } catch (IOException e) {
                            e.printStackTrace();
                        }
                        if (!udpSocket.isClosed()) {
                            udpSocket.close();
                        }
                        if (!service.isShutdown()) {
                            service.shutdown();
                        }
                    }
                });

                // if udp connection was able to connect with the port,
                // then set the state of the connection to true
                if (udpSocket.isConnected()) {

                    mavlinkConnection = MavlinkConnection.create(udpIn, udpOut);
                    mavlinkmessage = null;

                    while((mavlinkmessage = mavlinkConnection.next()) != null){
                        
                        if (message.getPayload() instanceof Heartbeat heartbeat) {}
                    }

                } else {
                    udpSocket.close();
                    Log.d("CONNECTION", "SOCKET FAILED TO CONNECT WITH MAVLINK");
                    return;
                }

                // closing connection : because we are out of loop at this point
                udpSocket.close();

            } catch (IOException e) {

                Log.e("Connection", "socket timeout: UDP mavlink service " + e.getMessage());
                mavlinkConnection = null;
            }
            
            
    }

as0919241, 2025-03-02:
// ConnectionDataClass holding the IP & PORT of connection
    ConnectionData udpDataRef = getSocketData();
    if (udpDataRef == null){
            return;
    }

        // The address we listen on to receive/send packets
        SocketAddress remoteAddress = new InetSocketAddress(udpDataRef.getIpAddress(), udpDataRef.getPort());

        // The size we use for buffers. 65535 should be able to carry any
        // UDP packet of a standard length.
        int bufferSize = 65535;

        try {
            // This binds to the listen address, we can later use this datagram socket
            // to receive packets and mark packets with source addresses when sending.
            DatagramSocket udpSocket = new DatagramSocket();
            udpSocket.connect(remoteAddress);

            PipedInputStream udpIn = new PipedInputStream();

            OutputStream udpOut = new OutputStream() {

                final byte[] buffer = new byte[bufferSize];
                int position = 0;

                @Override
                public void write(int b) throws IOException {

                    write(new byte[]{(byte) b}, 0, 1);
                }

                @Override
                public void write(byte[] b, int off, int len) throws IOException {

                    // if the buffer is full, we flush
                    if ((position + len) > (buffer.length)) {
                        flush();
                    }
                    System.arraycopy(b, off, buffer, position, len);
                    position += len;
                }

                @Override
                public void flush() throws IOException {
                    DatagramPacket packet = new DatagramPacket(buffer, 0, position, remoteAddress);
                    udpSocket.send(packet);
                    position = 0;
                }
            };

            // We connect `udpIn` to a corresponding PipedOutputStream (`appOut`). Data that we write to
            // `appOut` will be available to read from `udpIn`.
            // We instantiate/connect here rather than within the reading thread so that `udpIn` can be used immediately.
            PipedOutputStream appOut = new PipedOutputStream(udpIn);

            ExecutorService service = Executors.newSingleThreadExecutor();
            service.execute(() -> {
                try {
                    DatagramPacket packet = new DatagramPacket(new byte[bufferSize], bufferSize);
                    while (!udpSocket.isClosed()) {
                        udpSocket.receive(packet);
                        appOut.write(packet.getData(), packet.getOffset(), packet.getLength());
                        appOut.flush();
                    }
                } catch (IOException e) {
                    e.printStackTrace();
                } finally {
                    try {
                        appOut.close();
[... 39 lines cut ...]
        
}

This code is working fine. When we connect for the first time, both data sending and receiving work properly. However, if we turn the phone's WiFi off and then back on while staying on the same network, it reconnects, and data reception starts working again. But when we try to send a command, the data is not sent, and an exception occurs.

E  IOException while sending data: sendto failed: ENETUNREACH (Network is unreachable)
   java.io.IOException: sendto failed: ENETUNREACH (Network is unreachable)
       at libcore.io.IoBridge.maybeThrowAfterSendto(IoBridge.java:500)
       at libcore.io.IoBridge.sendto(IoBridge.java:688)
       at java.net.PlainDatagramSocketImpl.send(PlainDatagramSocketImpl.java:200)
       at java.net.DatagramSocket.send(DatagramSocket.java:500)
       at com.johnnette.gcsapp.Services.MavlinkService$2.lambda$sendData$0(MavlinkService.java:700)
       at com.johnnette.gcsapp.Services.MavlinkService$2.$r8$lambda$fEqVXYrU_WOEczFFfsqbUsNn5EU(Unknown Source:0)
       at com.johnnette.gcsapp.Services.MavlinkService$2$$ExternalSyntheticLambda0.run(Unknown Source:8)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
       at java.lang.Thread.run(Thread.java:1012)
   Caused by: android.system.ErrnoException: sendto failed: ENETUNREACH (Network is unreachable)
       at libcore.io.Linux.sendtoBytes(Native Method)
       at libcore.io.Linux.sendto(Linux.java:235)
       at libcore.io.ForwardingOs.sendto(ForwardingOs.java:600)
       at libcore.io.BlockGuardOs.sendto(BlockGuardOs.java:500)
       at libcore.io.ForwardingOs.sendto(ForwardingOs.java:800)

The exception occurs in the flush() method when calling udpSocket.send(packet);. The error "sendto failed: ENETUNREACH (Network is unreachable)"

oldGoodSee, 2025-05-21:
这是我的代码，目前可以通过 new InetSocketAddress(ip, port); 发送数据，new InetSocketAddress(localPort); 接收数据；其他socket等链接关闭问题需要自己去处理
`SocketAddress localAddress = new InetSocketAddress(localPort);
        SocketAddress remoteAddress = new InetSocketAddress(ip, port);
        int bufferSize = 65535;
        // 开启本地端口进行接收udp消息  todo 需要关闭
        DatagramSocket readSocket = new DatagramSocket(localAddress);
        // 用于发送udp消息   todo 需要关闭
        DatagramSocket sendSocket = new DatagramSocket();

        PipedInputStream udpIn = new PipedInputStream();
        OutputStream udpOut = new OutputStream() {
            final byte[] buffer = new byte[bufferSize];
            int position = 0;

            @Override
            public void write(int i) throws IOException {
                write(new byte[]{(byte) i}, 0, 1);
            }

            @Override
            public synchronized void write(byte[] b, int off, int len) throws IOException {
                // If the buffer is full, we flush
                if ((position + len) > buffer.length) {
                    flush();
                }
                System.arraycopy(b, off, buffer, position, len);
                position += len;
            }

            @Override
            public synchronized void flush() throws IOException {
                DatagramPacket packet = new DatagramPacket(buffer, 0, position, remoteAddress);
                sendSocket.send(packet);
                position = 0;
            }
        };

        // 通过udp接收到数据放入到 udpIn中 用mavlink协议去解析数据
        customExecutor.execute(() -> {
            try (PipedOutputStream appOut = new PipedOutputStream(udpIn)) {
                DatagramPacket packet = new DatagramPacket(new byte[bufferSize], bufferSize);
                while (!readSocket.isClosed()) {
                    readSocket.receive(packet);
                    appOut.write(packet.getData(), packet.getOffset(), packet.getLength());
                    appOut.flush();
                }
            } catch (IOException e) {
                logger.error("接收udp消息失败", e);
            }
        });

        return MavlinkConnection.create(udpIn, udpOut);`