dronefleet/mavlink #36: Payload length is limited to 253

iSorp, 2019-10-26:
When serializing (ReflectionPayloadSerializer) an FILE_TRANSFER_PROTOCOL message the exception "IllegalStateException: payload length > 253 for messageio.dronefleet.mavlink.common.FileTransferProtocol" is thrown but the FILE_TRANSFER_PROTOCOL has a length of 254. Shouldn't have a payload a maximum length of 255 bytes?

benbarkay, 2019-10-26:
Seeing as I can't recall why this limit is there in the first place, I think you're right. Unfortunately it might take me a little while (a week or two) to make a release with a fix for this.

aarongnielsen, 2020-03-30:
Hi Ben,
I suspect the error here was actually in the MAVLink developer documentation.
According to [this issue](https://github.com/mavlink/mavlink-devguide/issues/91), the documentation used to include the target system ID and target component ID as separate fields from the payload. This is incorrect: it depends on each message type whether these two IDs are included, and if they are, they appear wherever the serialiser puts them, based on the lengths of other fields. The correction is discussed [here](https://github.com/mavlink/mavlink-devguide/pull/100).
The fix for the Dronefleet library appears to be as simple as changing the constant in the serialiser from 253 bytes to 255. I can't see any other reason for the 253-byte limit either. In my own testing, I changed the annotation of the FileTransferProtocol.payload() method to limit the field from 251 bytes to 250, and this was enough to get a file transfer to work with a PX4 SITL.
Hope that helps. :)
Cheers,
Aaron

commit 9d93ce8c (2020-04-02): fix: #36 increased payload length from 253 to 255
closed 2020-04-02 by benbarkay