dronefleet/mavlink #38: SetMode.java Not Serializing Custom Modes Correctly For PX4

jameslabundis, 2019-11-26:
Background:
In attempting to use PX4's  Follow Me mode I found that the serialization for the CMD_DO_SET_MODE command does not account for the latest PX4 build and how it handles setting modes through Mavlink, at least when trying to use SetMode.java to do this. I'm assuming that SetMode is a wrapper than contains fields for the CMD_DO_SET_MODE. With a BaseMode field and a CustomMode.

How I found this Issue:

I sent a SetMode with a CustomMode of 12 for the FollowMe custom mode supported by PX4 and BaseMode of MAV_MODE_AUTO_ARMED. Despite setting what I thought were the right params the mode was not being set correctly in my simulation environment (JMAVSIM). 

What went wrong:

SetMode.java does not add a custom mode flag even when I specify a CustomMode is set. By not adding this, PX4 just uses the BaseMode I provided in SetMode instance. Secondly, PX4 does not just use one single CustomMode. PX4 uses a main CustomMode, then based off of that main CustomMode PX4 allows a subset of Sub CustomModes to be set.

PX4 Firmware (Reference for handling of MAV_CMD_DO_SET_MODE):
https://github.com/PX4/Firmware/blob/a8ea55d9b63df3ef2a6b688b7dd1ae107916a8a6/src/modules/commander/Commander.cpp

PX4 Custom Mode Enums:
https://github.com/PX4/Firmware/blob/07d656e971a72d1202651dfd3b4642736fb078d7/src/modules/commander/px4_custom_mode.h

How I got around this:

Using a CommandLong containing MAV_CMD_DO_SET_MODE with the parameters: BaseMode of 221 (Doesn't matter what is put here PX4 only checks the lowest bit for a one), 4 (PX4_CUSTOM_MAIN_MODE_AUTO), and 8 (PX4_CUSTOM_SUB_MODE_AUTO_FOLLOW_TARGET).

Suggestions: 
The PX4 documentation wasn't very clear one what I needed to pass to MAV_CMD_DO_SET_MODE. The parameter list suggested 12, but that's not how the firmware works. The firmware is hierarchical: 1st get set the right main CustomMode then the correct sub CustomMode. I'd suggest just making SetMode.java account for this.

ben-crowhurst, 2019-11-26:
@jameslabundis could you supply a PR and test cases replicating/validating the issue?

benbarkay, 2019-11-26:
If I can understand this correctly, you are looking to support the PX4-specific modes that are listed in the linked repositories. (Correct me if I'm wrong)

Specific support for these shouldn't be part of this library, as this library is only concerned with creating an API for the dialects. It would be appropriate to support this in a higher level codebase that uses dronefleet/mavlink -- Perhaps with a bunch of other stuff like abstraction which would make the same API compatible across different firmwares/versions.

I was planning on this with dronefleet/rxmavlink (making one setMode that would support multiple vehicles/versions), but as of now, I don't know when this will be released if at all.

closed 2020-03-30 by benbarkay