dronefleet/mavlink #41: How to write own dialect XML file

ImranGA, 2020-03-05:


benbarkay, 2020-03-05:
With the current project structure of the project, you have to clone the
repository, add your own dialect file to the dialects directory, run the
`generateDialects` gradle task, and run the gradle build.

ImranGA, 2020-03-05:
Thanks for it. Could you please tell me how to run `generateDialects` gradle task, and run the gradle build as I am new using gradle.

ImranGA, 2020-03-05:
[image]
Hi benbarkay I am getting this error. Please suggest me how to solve. Thank you

benbarkay, 2020-03-05:
If your default Java version is lower than 1.8 then you should upgrade,
it's probably failing because of the recent TLSv1.2 requirement from Maven
Central.
Otherwise it could be the gradle version that the project uses, however I
just managed to retrieve the dependency and run the build on this machine
using exactly whats on github (gradle 3.3)...

ImranGA, 2020-03-05:
Thanks issue resolved

closed 2020-03-05 by ImranGA