dronefleet/mavlink #49: Deserialization performance

m-sasha, 2020-08-05:
Deserialization performance can be drastically improved by caching the list of builder methods. Here's a typical flame chart for reading mavlink messages in a loop:
[image]

You can see that sorting the builder methods takes almost half the deserialization time. This can easily be solved with:

```
    private static final Map<Class<?>, List<Method>> builderMethodsCache = Collections.synchronizedMap(new HashMap<>());

    private static List<Method> getBuilderMethods(Class<?> builderClass){
        return builderMethodsCache.computeIfAbsent(builderClass, (c) -> Arrays.stream(c.getMethods())
                .filter(m -> m.isAnnotationPresent(MavlinkFieldInfo.class))
                .sorted((a, b) -> {
                    MavlinkFieldInfo fa = a.getAnnotation(MavlinkFieldInfo.class);
                    MavlinkFieldInfo fb = b.getAnnotation(MavlinkFieldInfo.class);
                    return wireComparator.compare(fa, fb);
                }).collect(Collectors.toList()));
    }
```

in `ReflectionPayloadDeserializer`.

benbarkay, 2020-08-05:
That's a good idea. If you make a PR then I'll merge it and release over the weekend.

m-sasha, 2020-08-06:
To further improve performance, I modified `MessageGenerator` to add to each type a `deserialize` method which is directly aware of its own fields and can deserialize the type without reflection. E.g.

```
    public static Heartbeat deserialize(ByteBuffer input) {
        long customMode = SerializationHelper.decodeUint32(input);
        EnumValue<MavType> type = SerializationHelper.decodeEnum(io.dronefleet.mavlink.common.MavType.class, input, 1);
        EnumValue<MavAutopilot> autopilot = SerializationHelper.decodeEnum(io.dronefleet.mavlink.common.MavAutopilot.class, input, 1);
        EnumValue<MavModeFlag> baseMode = SerializationHelper.decodeEnum(io.dronefleet.mavlink.common.MavModeFlag.class, input, 1);
        EnumValue<MavState> systemStatus = SerializationHelper.decodeEnum(io.dronefleet.mavlink.common.MavState.class, input, 1);
        int mavlinkVersion = SerializationHelper.decodeUint8(input);
        return new Heartbeat(type, autopilot, baseMode, customMode, systemStatus, mavlinkVersion);
    }
```

I've yet to test this, but judging from the flame chart, it should improve performance by at least 10x.
Would you like me to create a PR for this too?

MohammadAdib, 2022-01-20:
> To further improve performance, I modified `MessageGenerator` to add to each type a `deserialize` method which is directly aware of its own fields and can deserialize the type without reflection. [...]

Did you ever get around to making a PR for this?

m-sasha, 2022-01-20:
No, but if you want to continue where I left off, here's the patch:
[[WIP]_Direct_serialization.patch.zip](https://github.com/dronefleet/mavlink/files/7903326/WIP._Direct_serialization.patch.zip)