dronefleet/mavlink #51: Deserialization failure

franshua, 2020-10-20:
I have detected a pair of problems related with deserialization of ReflectionPayloadDeserializer. It does not convert payload to object value param:

- With the MemoryVect event, the packet received is:

MavlinkPacket{versionMarker=253, incompatibleFlags=0, compatibleFlags=0, sequence=48, systemId=1, componentId=1, messageId=249, payload=[1, 0, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1, 0, 1], checksum=35422, signature=[]}

and the final MemoryVect object is MemoryVect{address=1, ver=0, type=1, **value=null**}.

- With the BatteryStatus event is the same behaviour with voltages param:

MavlinkPacket{versionMarker=253, incompatibleFlags=0, compatibleFlags=0, sequence=86, systemId=1, componentId=1, messageId=147, payload=[1, 0, 0, 0, 1, 0, 0, 0, 25, 0, 4, 0, 4, 0, 4, 0, 4, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 1, 1, 1, 1, 48, 117, 0, 0, 1], checksum=43340, signature=[]}

BatteryStatus{id=1, batteryFunction=EnumValue{value=1, entry=MAV_BATTERY_FUNCTION_ALL}, type=EnumValue{value=1, entry=MAV_BATTERY_TYPE_LIPO}, temperature=25, **voltages=null**, currentBattery=1, currentConsumed=1, energyConsumed=1, batteryRemaining=1, timeRemaining=30000, chargeState=EnumValue{value=1, entry=MAV_BATTERY_CHARGE_STATE_OK}

Any idea about what could be happening? Thanks in advance.

benbarkay, 2021-01-12:
Sorry that it took me so long to respond, this has been fixed in master for a while but I could not find the time to make a proper test and release it.

The issue is with deserialization of arrays that are not `byte[]` (the version in master will deserialize them to `List`).

benbarkay, 2021-04-15:
Finally, `1.1.9` has been released with the fix.

closed 2021-04-15 by benbarkay