dronefleet/mavlink #53: [General guidance] Documentation for using the generator gradle plugin

Drkstr, 2021-01-11:
Hi,
      I have the MavLink project(`io.dronefleet.mavlink:mavlink:1.1.8`) imported into my Android project but I am not seeing class for any of the messages. Guessing I need to generate these using the generator gradle plugin. 

Do you have any documentation or examples on how I should use the plugin?

benbarkay, 2021-01-12:
There aren't such classes. The XML processing happens at build-time, and it generates annotated classes from it such as `Heartbeat`, which you can then use to build message objects that can be serialized or deserialized via `ReflectionPayloadSerializer` or `ReflectionPayloadDeserializer`.

If you want a custom message, then you have 2 options at the moment:
1. You could clone the repository, edit the XML files in `/definition-xml`, and then run the `generateDialect` gradle task; or
2. You could create your own `MavlinkDialect` implementation, depending on existing dialects (such as `CommonDialect`, for instance), and adding your own annotated message classes.

If you want to go for option 2, you can take a look at [UavionixDialect.java](https://github.com/dronefleet/mavlink/blob/master/src/main/java-gen/io/dronefleet/mavlink/uavionix/UavionixDialect.java) which depends on `CommonDialect` and adds some of its own messages.

Drkstr, 2021-01-13:
Thanks for your reply, 

So I just do a gradle build to generate the common message parsing classes?

closed 2021-01-29 by Drkstr