dronefleet/mavlink #56: I connect via usb serial, I get events (onreceive(byte[] data)) back, no inputstream or outputstream

MichaelSwan3, 2021-02-17:
I connect via usb serail on android, and I don't have a stream to use.

The data gets to a receive function:
```
    private void receive(byte[] data) {
        createMavLinkMessage(data);
```
...

This feels like a real clunky way of doing it. Is there a better way ?

```
 void createMavLinkMessage(byte[] data) {
        
        // It's difficult to create a continuous data stream, so I create 1 stream at a time and process the data 
        // all at once. Then I create a new stream and do it again.
        ByteArrayInputStream serialStream = new ByteArrayInputStream(data);

        OutputStream fakeOutputStream = null; // new OutputStream();
        MavlinkConnection connection = MavlinkConnection.create(serialStream, fakeOutputStream);

        try {
            MavlinkMessage message;
            while ((message = connection.next()) != null) {
                onNewMessage(message);
            }
        }  catch (EOFException eof) {
            Log.e(TAG, "End of Stream", eof);
            // onConnectionClose();
        } catch (IOException ioe) {
            Log.e(TAG, "Failed to establish Stream", ioe);
            // onConnectionClose();
        }
    }

```

benbarkay, 2021-02-18:
Take a look at `PipedInputStream` and `PipedOutputStream`, these are more
suitable for your situation

MichaelSwan3, 2021-02-22:
Thanks Ben,

That worked great! eg

` // Data Flowing from Drone To Android
    PipedOutputStream droneToAndroidOutput;
    PipedInputStream droneToAndroidInput;

    // Data Flowing from Android To Drone (requests etc)
    PipedOutputStream androidToDroneOutput;
    PipedInputStream androidToDroneInput;

...

  try {
            // Drone to tablet
            droneToAndroidOutput = new PipedOutputStream();
            droneToAndroidInput = new PipedInputStream(droneToAndroidOutput);

            // tablet to drone
            androidToDroneOutput = new PipedOutputStream();
            androidToDroneInput = new PipedInputStream(androidToDroneOutput);
        } catch(Exception e) {
            Log.e(TAG, "can't create piped stream" + e);
        }

... 

connection = MavlinkConnection.create(droneToAndroidInput, androidToDroneOutput);

/**
     * Data - data Frone the drone to the android tablet
     * @param data
     */
    public void addToPipe(byte[] data) {
        try {
            droneToAndroidOutput.write(data);
        } catch (IOException e) {
            Log.e(TAG, "Can't write piped stream" + e);
        }
    }

public void WriteHeartbeat() {
        int systemId = 255;
        int componentId = 0;
        Heartbeat heartbeat = Heartbeat.builder()
                .type(MavType.MAV_TYPE_GCS)
                .autopilot(MavAutopilot.MAV_AUTOPILOT_INVALID)
                .systemStatus(MavState.MAV_STATE_UNINIT)
                .mavlinkVersion(3)
                .build();

        // Write an unsigned heartbeat
        try {
            connection.send2(systemId, componentId, heartbeat);
        } catch (IOException e) {
            Log.e(TAG, "issue sending heartbeat to drone", e);
        }
    }
`

Cheers!

closed 2021-04-15 by benbarkay