dronefleet/mavlink #58: No such file or directory  #include "common/mavlink.h"

TurboTu, 2021-03-16:
Hi,
I tried to add some xmls into directory 'definition-xml' and run task 'generateDialect'.And the error occurred when I build the 'mavlink-1.1.8.jar'.It shows No such file or directory  #include "common/mavlink.h".Is there any idea to solve this?
Thanks.
[image]

benbarkay, 2021-03-16:
Try running `git submodule update --init --recursive` in your project,
The tests depend on the reference mavlink code from mavlink/mavlink,
It seems to be missing.

TurboTu, 2021-03-16:
@benbarkay 
Thanks a lot, I pulled again and it works.My platform is Windows10, and got a new error below.
[image]

benbarkay, 2021-03-16:
I'm not sure how to help you resolve this as I don't have access to a Windows machine.

Perhaps you should simply remove these tests. You can start by removing the test tool project from `settings.gradle` and then continue removing everything that's emitting an error when you build.

TurboTu, 2021-03-16:
Thanks!
It works as I commented lines in 'settings.gradle' and 'build.gradle'.

closed 2021-03-16 by TurboTu