dronefleet/mavlink #59: Battery voltage list causes a crash when adding it to the builder

MichaelSwan3, 2021-04-14:
Hi,

I'm trying to create my own batterystatus messages for testing. However it crashes everytime I create a voltage list.

Also, I don't know if this is related, but when I connected to the simulated drone in mission planner, voltages list is always null.

Creating voltages in BatteryStatus
 ```
 public MavMsg<BatteryStatus> createBatteryStatus(int id, int voltage, int batteryRemaining, int currentBattery) {

        /*
        // add 10 cells
        List<Integer> voltages = new ArrayList<Integer>();
        voltages.add(voltage);
        voltages.add(65535);
        voltages.add(65535);
        voltages.add(65535);
        voltages.add(65535);
        voltages.add(65535);

        voltages.add(65535);
        voltages.add(65535);
        voltages.add(65535);
        voltages.add(65535);
*/

        BatteryStatus batteryStatus = BatteryStatus.builder()
                .batteryRemaining(batteryRemaining)
                .currentBattery(currentBattery)
                .id(id)
               // .voltages(voltages)

                .build();

        // crashes here if we try to add voltage
        MavMsg<BatteryStatus> mmbat = new MavMsg<BatteryStatus>(batteryStatus);
```

Mission planner battery setup
[image]

Voltages are always null
[image]

It makes me think that there could be an issue with voltages in BatteryStatus? Most likely I've done something wrong though.

Cheers

MichaelSwan3, 2021-04-14:
Specifically it crashes here:
[image]

[image]

MichaelSwan3, 2021-04-14:
It crashes when it tries to convert it into bytes

[image]

[image]

So, it doesn't have a way of converting List<integer> into a set of bytes.

MichaelSwan3, 2021-04-14:
It looks like it might be fixed in here.
https://github.com/dronefleet/mavlink/commit/2255e6df02974af2031c05dcb509309ce75bdb5e

We are using the version from here

> Get it
> Maven Central coordinates are io.dronefleet.mavlink:mavlink:1.1.8

Has this been updated with the latest changes and if not, is it possible to get it updated ?

benbarkay, 2021-04-14:
This fix was indeed present in `master`.
I've released version 1.1.9. Should be available in central shortly.

closed 2021-04-14 by benbarkay

MichaelSwan3, 2021-04-16:
Ben,
I've got the new version and so far it's working great - 
Thanks!