dronefleet/mavlink #64: issue - does not have a builder

OctavianIonel, 2021-07-15:
I have a small java project where I included the forked generator repo as git submodule. In the same project I have another git submodule where I place some XML.
I bundle this project as a JAR in an android project, but at runtime I get this error at runtime

2021-07-15 18:19:11.672 16470-16669/com.mycompany.androidapp.develop E/AndroidRuntime: FATAL EXCEPTION: Thread-7
    Process: com.mycompany.androidapp.develop, PID: 16470
    io.dronefleet.mavlink.serialization.MavlinkSerializationException: Message com.mycompany.transmission.mavlink.common.Timesync does not have a builder
        at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.lambda$deserialize$1(ReflectionPayloadDeserializer.java:38)
        at io.dronefleet.mavlink.serialization.payload.reflection.-$$Lambda$ReflectionPayloadDeserializer$T4j64A0YnYVb-FB-1KA0v9Ei_Pc.get(Unknown Source:2)
        at java.util.Optional.orElseThrow(Optional.java:286)
        at io.dronefleet.mavlink.serialization.payload.reflection.ReflectionPayloadDeserializer.deserialize(ReflectionPayloadDeserializer.java:37)
        at io.dronefleet.mavlink.MavlinkConnection.next(MavlinkConnection.java:243)
        at com.mycompany.transmission.TransmissionAPI.lambda$receiveMessage$0(TransmissionAPI.java:48)
        at com.mycompany.transmission.TransmissionAPI.lambda$receiveMessage$0$TransmissionAPI(Unknown Source:0)
        at com.mycompany.transmission.-$$Lambda$TransmissionAPI$YnCdZ5iu-a4rPmBn3rc_WVPqv-k.run(Unknown Source:2)
        at java.lang.Thread.run(Thread.java:919)
2021-07-15 18:19:11.824 16470-16575/com.mycompany.androidapp.develop V/FA: Connecting to remote service

In the library (the one that is bundled as jar I have the following dependency:

    <dependency>
      <groupId>io.dronefleet.mavlink</groupId>
      <artifactId>mavlink</artifactId>
      <version>1.1.9</version>
    </dependency>

And also in the Android project I have:

    implementation 'io.dronefleet.mavlink:mavlink:1.1.9'

OctavianIonel, 2021-07-19:
@benbarkay @h4writer @lmar @m-sasha 
Could you help me please?

benbarkay, 2021-07-19:
Could you include the source for:
com.mycompany.transmission.mavlink.common.Timesyn ?

OctavianIonel, 2021-07-19:
Nevertheless, I get the same error also with other types, probably based on which is arriving first (another error is complaining about this: RadioStatus does not have a builder)

```
package com.mycompany.transmission.mavlink.common;
import io.dronefleet.mavlink.annotations.MavlinkFieldInfo;
import io.dronefleet.mavlink.annotations.MavlinkMessageBuilder;
import io.dronefleet.mavlink.annotations.MavlinkMessageInfo;
import java.lang.Object;
import java.lang.Override;
import java.lang.String;
import java.util.Objects;

/**
 * Time synchronization message. 
 */
@MavlinkMessageInfo(
        id = 111,
        crc = 34,
        description = "Time synchronization message."
)
public final class Timesync {
    private final long tc1;

    private final long ts1;

    private Timesync(long tc1, long ts1) {
        this.tc1 = tc1;
        this.ts1 = ts1;
    }

    /**
     * Returns a builder instance for this message.
     */
    @MavlinkMessageBuilder
    public static Builder builder() {
        return new Builder();
    }

    /**
     * Time sync timestamp 1 
     */
    @MavlinkFieldInfo(
            position = 1,
            unitSize = 8,
            signed = true,
            description = "Time sync timestamp 1"
    )
    public final long tc1() {
        return this.tc1;
    }

    /**
     * Time sync timestamp 2 
     */
    @MavlinkFieldInfo(
            position = 2,
            unitSize = 8,
            signed = true,
            description = "Time sync timestamp 2"
    )
    public final long ts1() {
        return this.ts1;
    }

    @Override
    public boolean equals(Object o) {
        if (this == o) return true;
        if (o == null || !getClass().equals(o.getClass())) return false;
        Timesync other = (Timesync)o;
        if (!Objects.deepEquals(tc1, other.tc1)) return false;
        if (!Objects.deepEquals(ts1, other.ts1)) return false;
        return true;
    }

    @Override
    public int hashCode() {
        int result = 0;
        result = 31 * result + Objects.hashCode(tc1);
        result = 31 * result + Objects.hashCode(ts1);
        return result;
    }

    @Override
    public String toString() {
        return "Timesync{tc1=" + tc1
                 + ", ts1=" + ts1 + "}";
    }

    public static final class Builder {
        private long tc1;

        private long ts1;

        /**
         * Time sync timestamp 1 
         */
        @MavlinkFieldInfo(
                position = 1,
                unitSize = 8,
                signed = true,
                description = "Time sync timestamp 1"
        )
        public final Builder tc1(long tc1) {
            this.tc1 = tc1;
            return this;
        }

        /**
         * Time sync timestamp 2 
         */
        @MavlinkFieldInfo(
                position = 2,
                unitSize = 8,
                signed = true,
                description = "Time sync timestamp 2"
        )
        public final Builder ts1(long ts1) {
            this.ts1 = ts1;
            return this;
        }

        public final Timesync build() {
            return new Timesync(tc1, ts1);
        }
    }
}
```

benbarkay, 2021-07-19:
The class seems to have been generated OK.

Try disabling ProGuard in your build and see if this is still happening.

OctavianIonel, 2021-07-19:
That was it...OMG, but why this should happen? It gets confused with the references?

OctavianIonel, 2021-07-19:
Many thanks @benbarkay !!

closed 2021-07-19 by benbarkay

OctavianIonel, 2021-07-19:
I have added a rule in proguard in order not to obfuscate the code regarding Mavlink.
Many thanks!