dronefleet/mavlink #70: system not showing any result

zeev-mindali, 2022-02-05:
Dear @benbarkay 

i trying to use this simple code (with mavsdk_server):
```
   Socket socket = new Socket("localhost", 50051);
        MavlinkConnection connection = MavlinkConnection.builder(socket.getInputStream(),
                        socket.getOutputStream())
                .dialect(MavAutopilot.MAV_AUTOPILOT_ARDUPILOTMEGA, new ArdupilotmegaDialect())
                .build();

        CommandLong setPWM = CommandLong.builder()
                .command(MavCmd.MAV_CMD_DO_SET_SERVO)
                .confirmation(0)
                .param1(4).param2(1200).param3(0).param4(0).param5(0).param6(0).param7(0)
                .targetSystem(1).targetComponent(1)
                .build();

        connection.send2(255,0,setPWM);
```

don't see any result, what am i missing here?

benbarkay, 2022-02-06:
If the connection succeeded, `send2` has been executed, and the JVM isn't exiting, then this message should be delivered to the other end at 50051.

How come you're not reading input from the connection?
It would be a lot easier to debug if you could see a response message.

Try adding this after the last `send2`:

```java
MavlinkMessage message;
while ((message = connection.next()) != null) {
    System.out.println(message);
}
```

zeev-mindali, 2022-02-06:
@benbarkay 

add as you told

mavproxy :
```
zeevmindali@zeevs-MacBook-Pro sitl % ./runserver.sh
[12:37:22|Info ] MAVSDK version: v1.0.8 (mavsdk_impl.cpp:28)
[12:37:22|Info ] Waiting to discover system on udp://192.168.1.15:14552... (connection_initiator.h:20)
[12:37:22|Debug] Initializing connection to remote system... (mavsdk_impl.cpp:482)
[12:37:22|Debug] New: System ID: 42 Comp ID: 251 (mavsdk_impl.cpp:220)
[12:37:22|Debug] Component Unsupported component (251) added. (system_impl.cpp:360)
[12:37:22|Debug] New: System ID: 255 Comp ID: 190 (mavsdk_impl.cpp:484)
[12:37:22|Debug] Component Unsupported component (190) added. (system_impl.cpp:360)
[12:37:22|Debug] Discovered 1 component(s) (system_impl.cpp:528)
[12:37:22|Info ] System discovered (connection_initiator.h:63)
[12:37:22|Info ] Server started (grpc_server.cpp:52)
[12:37:22|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:53)
[12:37:22|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:484)
[12:37:22|Debug] Component Autopilot (1) added. (system_impl.cpp:360)
[12:37:22|Debug] Discovered 1 component(s) (system_impl.cpp:528)
[12:37:23|Debug] Component Camera 1 (100) added. (system_impl.cpp:360)
[12:37:23|Debug] MAVLink: critical: PreArm: Hardware safety switch (system_impl.cpp:239)
[12:37:23|Debug] MAVLink: critical: PreArm: Rangefinder 1 was not detected (system_impl.cpp:239)
[12:37:25|Debug] Component Unsupported component (250) added. (system_impl.cpp:360)
```

and the code in java:
```
 public static void main(String[] args) {
        MavlinkConnection connection = null;
        try {
            Socket socket = new Socket("127.0.0.1", 50051);
            connection = MavlinkConnection.builder(socket.getInputStream(),
                            socket.getOutputStream())
                    .dialect(MavAutopilot.MAV_AUTOPILOT_ARDUPILOTMEGA, new ArdupilotmegaDialect())
                    .build();
        } catch (IOException e) {
            e.printStackTrace();
        }

        try {
            connection.send2(255, 0, serServo(5, 1200));
            MavlinkMessage message;
            while ((message = connection.next()) != null) {
                System.out.println(message);
            }
        } catch (IOException e) {
            e.printStackTrace();
        }
    }

    private static CommandLong serServo(int channel, int pwm) {
        return CommandLong.builder()
                .command(MavCmd.MAV_CMD_DO_SET_SERVO)
                .confirmation(0)
                .param1(channel).param2(pwm).param3(0).param4(0).param5(0).param6(0).param7(0)
                .targetSystem(1).targetComponent(1)
                .build();
    }
```

zeev-mindali, 2022-02-06:
@benbarkay 
o.k., I run mavproxy udpout:14552 , coin :14550
getting finally the heartbeat :) cheers

but still, can not sending command as described above :(

zeev-mindali, 2022-02-06:
@benbarkay I need little push, understand why it's not sending and I have the system
please help

benbarkay, 2022-02-06:
I can't see anything wrong with this,

Perhaps you can try flushing the output stream after `send2`:
```java
socket.getOutputStream().flush();
```

If this doesn't work, perhaps try using some of the other commands to see if you can get a response for these.
If that works, then you may have to either switch firmware to one that's compatible with the dialect versions of the library or clone the repository and generate message classes based on the dialect XML files that match the firmware you're using.

zeev-mindali, 2022-02-20:
@benbarkay still no success :(

zeev-mindali, 2022-02-22:
@benbarkay please any help? stuck with my project :(
it's seems it's stuck on connection.next()

zeev-mindali, 2022-02-24:
ITS WORKING !!!!!!!

after digging and digging, it's was problem with JVM and UDP.
now I controlling the drone :)

closed 2022-02-27 by zeev-mindali

YeganehNobakht, 2024-09-17:
Dear @zeev-mindali 
I stuck in my project and the issue is similar yours. how did you solve your issue?
I try to send message via serial port. There is no problem with receiving data but sending command is not work.
this is my code:

CommandLong rtlMessage = CommandLong.builder()
            .targetSystem(1)
            .targetComponent(1)
            .command(MavCmd.MAV_CMD_NAV_RETURN_TO_LAUNCH)
            .param1(Float.NaN)
            .param2(Float.NaN)
            .param3(Float.NaN)
            .param4(Float.NaN)
            .param5(Float.NaN)
            .param6(Float.NaN)
            .param7(Float.NaN)
            .build();
        mavlinkConnection.send2(1,1, rtlMessage );