dronefleet/mavlink #85: MAV_CMD_DO_SET_MODE (176) 切换模式失败

KKJava1, 2025-06-06:
`COMMAND_LONG {target_system : 1, target_component : 1, command : 176, confirmation : 0, param1 :1 , param2 : 0x30000, param3 : 0.0, param4 : 0, param5 : 0.0, param6 : 0.0, param7 : 0.0}`

目前切换模式是通过这样的参数传递，但是我根据这样的参数进行传递后被拒绝

`
    public void setUavMode(String sn, Integer mode) throws IOException {
        float modeValue = 0;

        UavDataCacheManager cacheManager = UavDataCacheManager.getInstance();

        // 根据传入的模式值设置 modeValue
        switch (mode) {
            case 1:
                // 控制模式
                modeValue = 0x60000;
                break;
            case 2:
                // 定点模式
                modeValue = 0x30000;
                break;
            case 3:
                // 自动模式
                modeValue = 0x4000000;
                break;
            default:
                break;
        }

        // 判断传入的模式和当前模式是否相同
        boolean exists = cacheManager.hasUavData(sn);
        if (exists) {
            Map<String, Object> uavData = cacheManager.getUavDataBySn(sn);
            Object cachedModeNum = uavData.get("modeNum");
            if (cachedModeNum instanceof Float && (Float) cachedModeNum == modeValue) {
                return;
            }
        }

        // 发送命令设置模式
        sendCommand(
                sn,
                MavCmd.MAV_CMD_DO_SET_MODE,
                1,
                Float.parseFloat(Integer.toHexString((int) modeValue)),  // 直接传递十六进制值
                0, 0, 0, 0, 0);

        // 当前线程暂停 1 秒
        long countTime = 1000_000L * 1_000;
        LockSupport.parkNanos(countTime);
    }

    private void sendCommand(
            String sn,
            MavCmd command,
            float param1, float param2, float param3, float param4,
            float param5, float param6, float param7) throws IOException {
        CommandLong cmd = new CommandLong.Builder()
                .command(command)
                .targetSystem(getMavlinkUav(sn).getTargetSystem().getTargetSystem()) // 飞控系统ID
                .targetComponent(getMavlinkUav(sn).getTargetSystem().getTargetComponent()) // 组件ID
                .param1(param1)
                .param2(param2)
                .param3(param3)
                .param4(param4)
                .param5(param5)
                .param6(param6)
                .param7(param7)
                .build();
        MavlinkConnection connection = getMavlinkUav(sn).getMavlinkConnection();
        logger.info("发送的指令为{}",cmd);
        connection.send1(
                properties.getSystemId(),
                properties.getComponentId(),
                cmd);
    }

`

`
2025-06-06 10:23:46.966 - [MQTT Call: re4d3f6a7cc42f0aac8015a1d68913fecd][] INFO  MavlinkService -切换模式的值为393216.0
2025-06-06 10:23:46.967 - [MQTT Call: re4d3f6a7cc42f0aac8015a1d68913fecd][] INFO  .MavlinkService -发送的指令为CommandLong{targetSystem=6, targetComponent=1, command=EnumValue{value=176, entry=MAV_CMD_DO_SET_MODE}, confirmation=0, param1=1.0, param2=393216.0, param3=0.0, param4=0.0, param5=0.0, param6=0.0, param7=0.0}
2025-06-06 10:23:47.010 - [Thread-8][] ERROR com.yuefei.uav.mavlink.service.MavlinkService -指令被拒绝: MAV_RESULT_DENIED，FROM：MAV_CMD_DO_SET_MODE
`
最后报错的结果是这样

vladnov138, 2025-06-20:
Why are you using hexadecimal format? It seems like the project is no longer maintained, so I saw your PR and wanted to help. As I understand it, the issue is that switching to any other mode is not working.
I’ve used the same dronefleet mavlink library in my project with Ardupilot, and here’s a Kotlin example that successfully switches the flight mode. Maybe it will be helpful for you:
`fun setGuidedMode() {
        val guidedModeCommand = CommandLong.builder()
            .targetSystem(1)
            .targetComponent(1)
            .command(MavCmd.MAV_CMD_DO_SET_MODE)
            .confirmation(0)
            .param1(127F) // MAV_MODE_FLAG_CUSTOM_MODE_ENABLED
            .param2(4F)   // GUIDED
            .param3(0F)
            .param4(0F)
            .param5(0F)
            .param6(0F)
            .param7(0F)
            .build()
        sendCommand(guidedModeCommand)
    }`

KKJava1, 2025-06-20:
> Why are you using hexadecimal format? It seems like the project is no longer maintained, so I saw your PR and wanted to help. As I understand it, the issue is that switching to any other mode is not working. I’ve used the same dronefleet mavlink library in my project with Ardupilot, and here’s a Kotlin example that successfully switches the flight mode. Maybe it will be helpful for you:
> 
>         val guidedModeCommand = CommandLong.builder()
>             .targetSystem(1)
>             .targetComponent(1)
>             .command(MavCmd.MAV_CMD_DO_SET_MODE)
>             .confirmation(0)
>             .param1(127F) // MAV_MODE_FLAG_CUSTOM_MODE_ENABLED
>             .param2(4F)   // GUIDED
>             .param3(0F)
>             .param4(0F)
>             .param5(0F)
>             .param6(0F)
>             .param7(0F)
>             .build()
>         sendCommand(guidedModeCommand)
>     }

thanks, Beacause I use Acfly drone for using hexadecimal format,This flight control mode switching requires a specific flight stack, so it is more troublesome.Now I always solove this problem,but I have some question for drone,I hope we can communicate together，Could you leave you weChat? thanks

vladnov138, 2025-06-21:
I'm interested in communicating with you. Unfortunately, I don't use WeChat and it's a bit difficult for me to sign up since I'm not a Chinese citizen, but you can contact me in Discord. We can continue our conversation there or if you prefer WeChat, maybe you could help me with the QR code login so we can connect.
Discord: vladislavnovikov

> thanks, Beacause I use Acfly drone for using hexadecimal format,This flight control mode switching requires a specific flight stack, so it is more troublesome.Now I always solove this problem,but I [...]

KKJava1, 2025-06-22:
> I'm interested in communicating with you. Unfortunately, I don't use WeChat and it's a bit difficult for me to sign up since I'm not a Chinese citizen, but you can contact me in Discord. We can [...]

Ok，I already add you in discord，by the way，I have some question for you。
Now,Already receive setMode method and control uav move. However, the steering function is not currently available.
```
    public void droneControl(String sn, float x, float y, float h, float w) throws Exception {
        isConnection(sn);
        
        setOldUavMode(sn, 1);  // change offboard mode
        
        float currentYawRad = 0.0f;
        if (w != 0) {
            UavDataCacheManager cacheManager = UavDataCacheManager.getInstance();
            Map<String, Object> uavDataBySn = cacheManager.getUavDataBySn(sn);
            if (ObjectUtil.isNotEmpty(uavDataBySn) && ObjectUtil.isNotEmpty(uavDataBySn.get("yaw"))) {
                currentYawRad = (float) uavDataBySn.get("yaw");
            }
            float deltaYaw = (float) Math.toRadians(-w); // Convert degrees to radians
            currentYawRad += deltaYaw;
            currentYawRad = (float) ((currentYawRad + Math.PI) % (2 * Math.PI) - Math.PI);
        }
        MavlinkUav uav = getMavlinkUav(sn);

        SetPositionTargetLocalNed message = SetPositionTargetLocalNed.builder()
                .targetSystem(uav.getTargetSystem())
                .targetComponent(uav.getTargetComponent())
                .coordinateFrame(MavFrame.MAV_FRAME_BODY_FRD)
                .typeMask(PositionTargetTypemask.POSITION_TARGET_TYPEMASK_VX_IGNORE,
                        PositionTargetTypemask.POSITION_TARGET_TYPEMASK_VY_IGNORE,
                        PositionTargetTypemask.POSITION_TARGET_TYPEMASK_VZ_IGNORE,
                        PositionTargetTypemask.POSITION_TARGET_TYPEMASK_AX_IGNORE,
                        PositionTargetTypemask.POSITION_TARGET_TYPEMASK_AY_IGNORE,
                        PositionTargetTypemask.POSITION_TARGET_TYPEMASK_AZ_IGNORE,
                        PositionTargetTypemask.POSITION_TARGET_TYPEMASK_YAW_RATE_IGNORE)
                .x(x)
                .y(y)
                .z(-h)
                .vx(0.0f)
                .vy(0.0f)
                .vz(0.0f)
                .afx(0.0f)
                .afy(0.0f)
                .afz(0.0f)
                .yaw(currentYawRad)
                .yawRate(0.0f)
                .build();
        
        MavlinkConnection connection = uav.getMavlinkConnection();
        logger.info("The command sent by the command flight is", message);
        connection.send2(
                uav.getTargetSystem(),
                uav.getTargetComponent(),
                message);
    }

```
This method can be used to control the movement of the drone, including up, down, left, and right.But it cannot be redirected. The redirection log is as follows

`2025-06-13 16:44:16.373 - [MQTT Call: re62834ba80076e952617679e966aa23b3][] INFO  com.yuefei.uav.mavlink.service.MavlinkService -SetPositionTargetLocalNed{timeBootMs=0, targetSystem=6, targetComponent=1, coordinateFrame=EnumValue{value=12, entry=MAV_FRAME_BODY_FRD}, typeMask=EnumValue{value=2552, entry=null}, x=0.0, y=0.0, z=-0.0, vx=0.0, vy=0.0, vz=0.0, afx=0.0, afy=0.0, afz=0.0, yaw=-0.24230176, yawRate=0.0}`

Is there something wrong with the method?

closed 2025-07-04 by KKJava1